ArduPilot/MissionPlanner #1540: Overlay from KML/KMZ transparency adds

lvale, 2017-05-02:
When working in zones (NFZ) where several KML/KMZ files overlay each other the transparency is added with each layer, which tends to make the map completely opaque.

Either we have a way to disable overlays like NFZ or better would be to have a lighter transparency, and even better not having the different layers adding each other.

Tested with latest MP on Win10

[image]
[image]

meee1, 2017-06-02:
im not sure i really want to change this. as it does a good job of indicating that there is more than one in that area. to maintain safety and user awareness

lvale, 2017-06-02:
@meee1 You are aware that the map gets completely opaque ? I agree that NFZ should remain, but reduce the transparency to a maximum would help when areas cross each other.