ArduPilot/MissionPlanner #1542: Joystick not working

peter249, 2017-05-05:
#### Issue details
Since last update: joystick not found / connected.  It is connected and recognized by windows 10 creators edition .

Traced back through code to Joystick.cs line 408 : JoyStickByName()

Name passed in ok "T.Flight Hotas X",  SharpDX.driver.ProductName() ruturns "T.Flight Hotas X\0\0\0......"

I looked (not verified) into the SharpDX code (I am on VS2017 Professional now -- VS2015 uninstalled)  and think it is in their Utilities.cs -- PtrToStringUni() function (line 523) which just calls Microsoft Marshel.PtrToStringUni() function which I think is not working correctly.

I did throw a quick program together to verify its NOT missionplanner and the call to SharpDX still fails. 

As a side note: I did find a bug in missionplanner. Joystick.cs Line 1450:

your code: working = (int) (working/65.535) - range/2;

should be: working = (int) (working / 65.535) - 500; 

or for integer math:  working =  ((working * 1000) / 65535)) - 500;

This will fix joystick problems when RC.MIN != 1000 and RC.MAX != 2000 as range relied on this fact!

 
#### Version
1.3.47 and 1.3.48

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
quad

#### Hardware type
Navio2

#### Logs
N/A

peter249, 2017-05-08:
Started an Issue with SharpDX number 892 and Alexandre suggested to use the prerelease 4.00-ci145.

I used NuGet manager and upgraded my test program to the prerelease and it worked.

I then went to the MissionPlanner source and updated SharpDX and SharpDX.DirectInput and now MissionPlanner sees the Joystick.  It seems to be working fine with this NuGet upgrade.  I am back in business, but, would always want to use the official release of MissionPlanner.

I also found an pull request which ExusAltimus submitted which probably addresses the same bug.

Note: my cut and past has error should have been

for integer math: working = ((working * 1000) / 65535) - 500; 

brackets are important....

schabauerj, 2017-05-31:
THANK YOU for tracing this bug down - I stumbled over it during a presentation.  Now I know that updating just before a big show is never a good idea ...
Anyway, so how does you commit make its way to the main repo now?  I just updated my Mission Planner from 1.3.46 to 1.3.48 and the bug is still there.

meee1, 2017-05-31:
this should all be resolved in beta. the \0's where causing issues when selecting a joystick. the scaling is also fixed and now using a map() command

schabauerj, 2017-05-31:
Thank's again, I just upgraded to 1.3.48.3 and the joystick works again!

closed 2017-06-02 by meee1