ArduPilot/MissionPlanner #1543: Survey Grid disply problem on Windows 7 tablet

MattyPope, 2017-05-06:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

MattyPope, 2017-05-06:
[image]
I'd like to use a windows 7 tablet for its daylight readable display but I have an issue with the survey grid display obscuring the camera settings.  Any ideas?

meee1, 2017-05-06:
you have your tablet set at a screen scale of more than 100%. as a work around set it to 100%. I will need to fix the internal scalling

MattyPope, 2017-05-06:
thanks .  I'll need some new glasses but thats got it sorted.  Love your work

closed 2017-05-06 by MattyPope