ArduPilot/MissionPlanner #1544: About Adding New No-Fly Zones

mustaphos, 2017-05-06:
#### Issue details
Hi. I want to add custom no-fly zone like of the airport list in the Mission Planner.
Where can i add my no-fly zones as lat, lon, radius?
Thanks for your answers.

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[x] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2017-05-08:
This is not support yet I'm afraid.  It's on the to-do list however:
https://github.com/ArduPilot/ardupilot/issues/391
https://github.com/ArduPilot/ardupilot/issues/1056

rmackay9, 2017-05-08:
Sorry, I thought this was on the ardupilot issues list but I see this could be a request to customise the list that Mission Planner shows for informational purposes to the pilot.

meee1, 2017-05-31:
MP uses kml's to display the no fly zones. just place a kml inside the nofly directory

closed 2017-05-31 by meee1