ArduPilot/MissionPlanner #1545: cache folder error on Linux

AndKe, 2017-05-07:
Some MP users claim it's viable solution on other things than windows.
It appears to work via mono on Linux, (Ubuntu)  - but this error pops up every time:
- also - it does not quit - but runs after that.
if it's a temporary cache, it should be stored in /tmp  , otherwise somewhere in user config, like  ~/.config/missionplanner/cache

[image]

kikislater, 2017-05-07:
Still issue since October 2016 : 
https://github.com/ArduPilot/MissionPlanner/issues/1379

AndKe, 2017-05-07:
thanks that confirms my impression of MP.
I saw a user recently claim that it worked fine in mono, so I thought there were maybe some attempts in the past years to get it out of the windows quagmire.

kikislater, 2017-05-07:
It works with mono Linux but not as well as under windows. There is some bugs, you have to refresh panels by switching to different menu (example : if you don't have HUD in flight data, move to flight plan and revert back to flight data and you could get HUD or resize window). Some functions does not work as expected, so definitively could not use as a GCS on field under Linux.
LogAnalyser does not work. In that case, I use it only for flight plan. Qgroundcontrol should be better in the future : more cross platform GCS wit QT. 
We choose a prooven opensource platform but discover after using it that apm planner is not reliable (lack of functions) and we have to use a software (Mission Planner) under a non opensource operating system.

AndKe, 2017-05-08:
thanks for sharing .
I am usually using AP2 , often flying in valleys and varying altitude, drawing mission manually. I recently started to try out QGC for planning,   using mavproxy + AP2 for flight monitoring. 
I use AP2 for log analysis, it's kind of perfect, fast and smooth solution even for big logs (many hours, and much data)

meee1, 2017-06-02:
in current beta ive moved this folder if mono is detected.

closed 2017-06-02 by meee1

kikislater, 2017-06-05:
Got it working, thanks. 
Which mono version should we use ? 4.6 ?

AndKe, 2017-06-28:
nope, does not work, and why not use /tmp for that ?   

![screenshot from 2017-06-28 20-30-23](https://user-images.githubusercontent.com/1468842/27653927-
e53ca954-5c40-11e7-84cc-82cdb0c6c06e.png)

meee1, 2017-06-28:
@AndKe  are you using the latest beta?

AndKe, 2017-06-29:
@meee1 - downloaded it yesterday from http://firmware.ardupilot.org/Tools/MissionPlanner/MissionPlanner-latest.zip
the attempt on /usr/share tells me it must be recent, not sure it "latest" equals "beta", or "stable"

meee1, 2017-06-29:
nope, http://firmware.ap.ardupilot.org/Tools/MissionPlanner/beta/

AndKe, 2017-06-29:
Beta works fine - thanks.