ArduPilot/MissionPlanner #1547: Survey grid: Distance between lines is wrong

StephBMG, 2017-05-07:
#### Issue details
Survey grid: Distance between lines
#### Version
1.3.48, the distance is wrong (320m) - 1.3.32, the distance is right (480m)
[image]
[image]

sidelap:60%

meee1, 2017-05-09:
can you do a control-s and post the file it saves?

iskess, 2017-05-09:
If this turns out to be a verified bug, could you let us know which versions were affected? I need to go back and check my surveys for accuracy. 
Thank you.

StephBMG, 2017-05-10:
Sorry, my fault was not a bug: since V1.3.40, the "front facing camera" is not ticked by default, I did not see that, sorry

closed 2017-05-10 by StephBMG