ArduPilot/MissionPlanner #1549: Request: built in simulated throttle for SITL

iskess, 2017-05-09:
I use the built-in SITL to check my flight plan for errors. I don't need a joystick to check Auto flight, other than to initiate the Takeoff. If MP had a virtual joystick, or even just a button that will override the need for the throttle to be raised for takeoff, SITL could be used in the field without the need to carry and calibrate a joystick. 
Perhaps a function to map keyboard keys to joystick commands?

magicrub, 2017-05-10:
Just type in 
rc 3 2000

It raises channel 3 pwm input to 200us. However, in an auto mission you shouldn't need to do that. It should take off without it.

iskess, 2017-05-10:
I'm just using the built in SITL on the MP Simulation tab. Where would I type in this command? The Terminal?

meee1, 2017-05-31:
normally I would connect a joystick, to simulate the throttle etc. is that not an option?

iskess, 2017-05-31:
When I'm in the field and I want to test a change to the flight plan, I don't always have a joystick available. If all I need is a throttle for just a moment to launch, it seems silly to carry around a joystick when a single click in MP should do what I need. What about just adding a SITL Takeoff button just to get airborne in Auto mode?

magicrub, 2017-05-31:
When I run SITL I control the throttle with the command "rc 3 2000" for
full throttle, "rc 3 1000" for zero throttle. Do you have access to
MAVProxy and can type that in on your SITL?

iskess, 2017-05-31:
I'm just using MP's built in simulator. I don't run MAVProxy. I was hoping not to have to figure out MAVProxy just to Takeoff without a joystick available. If @meee1 declines to add this functionality, I will attempt to follow your suggestion. 

Maybe there could be an option to start SITL in a Loiter over home so no take off is required?

SSUAV, 2017-05-31:
Doesn't adding "Auto Takeoff" to the start of the mission achieve this. Is triggered when you change the mode to Auto in MP.

iskess, 2017-06-01:
No, to initiate a Takeoff the throttle must be raised.