ArduPilot/MissionPlanner #1551: Allow Clear Track Button to Also Clear Camera Trigger Symbols

iskess, 2017-05-11:
When flying or playing back a tlog of a mapping mission, a green camera icon is placed at each triggering point. These can eventually get so dense that it obscures the underlying map. Especially when flying a cross grid pattern, or re-flying an area that didn't work out the first time. I haven't found any way to clear these symbols other than restarting MP. There is already a Clear Track button, could it also serve to clear the camera symbols?
It might also be nice to make the camera symbols semi-transparent so the map underneath is not obscured.

meee1, 2017-06-02:
added in beta

iskess, 2017-07-09:
The camera symbols now clear with the track, thank you. 
I'll leave this issue open in case you intend to implement the other ideas:
1) Semi-transparent camera symbols so we can view the map under it.
2) The newest symbol appears a different color for a couple seconds so its identifiable in a sea of older green symbols.  
Feel free to close this issue if you want, I'm not attached.