ArduPilot/MissionPlanner #1553: Arch Linux Mono MissionPlanner does not start because System.Speech is not found

t2000, 2017-05-14:
Hello,

I was using MP for quite a while on my old Laptop running Gentoo. However now on Arch Linux the MP does not start due to the following error. I do not need any speech functionality but also did not find any configuration file to disable this feature so MP starts again.

So can you please tell me which package I am missing or how I can disable this feature, since I do not use it anyway. Thank you.
[mp-problem.txt](https://github.com/ArduPilot/MissionPlanner/files/999336/mp-problem.txt)

```
stefan@xps13 ~/copter/missionplanner-1.3.48 $ ./MissionPlanner.sh 
: No such file or directory: #!/bin/sh
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35 

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe

(mono:7331): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(mono:7331): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
INFO Program - ******************* Logging Configured *******************
INFO Program - 64bit os True, 64bit process True
bus type BUS_TYPE_I2C bus 2 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883_OLD 
bus type BUS_TYPE_SPI bus 8 address (i2c addr or spi CS) 1 devtype DEVTYPE_AK8963 
bus type BUS_TYPE_SPI bus 8 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D 
bus type BUS_TYPE_SPI bus 8 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D 
bus type BUS_TYPE_SPI bus 2 address (i2c addr or spi CS) 4 devtype DEVTYPE_AK8963 
bus type BUS_TYPE_SPI bus 2 address (i2c addr or spi CS) 4 devtype DEVTYPE_AK8963 
bus type BUS_TYPE_SPI bus 8 address (i2c addr or spi CS) 1 devtype 22 
bus type BUS_TYPE_SPI bus 8 address (i2c addr or spi CS) 3 devtype 17 
bus type BUS_TYPE_SPI bus 2 address (i2c addr or spi CS) 4 devtype 22 
bus type BUS_TYPE_SPI bus 8 address (i2c addr or spi CS) 1 devtype 36 
bus type BUS_TYPE_SPI bus 8 address (i2c addr or spi CS) 4 devtype 34 
bus type BUS_TYPE_SPI bus 2 address (i2c addr or spi CS) 4 devtype 36 
FATAL Program - Fatal app exception
System.TypeLoadException: Could not load type of field 'MissionPlanner.Utilities.Speech:_speechwindows' (1) due to: Could not load file or assembly 'System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies. assembly:System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35 type:<unknown type> member:<none>
  at (wrapper remoting-invoke-with-check) MissionPlanner.MainV2:.ctor ()
  at MissionPlanner.Program.Main (System.String[] args) [0x005ef] in <f3fa022404f84f6f87ecbf3f76f088bf>:0 
System.TypeLoadException: Could not load type of field 'MissionPlanner.Utilities.Speech:_speechwindows' (1) due to: Could not load file or assembly 'System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies. assembly:System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35 type:<unknown type> member:<none>
  at (wrapper remoting-invoke-with-check) MissionPlanner.MainV2:.ctor ()
  at MissionPlanner.Program.Main (System.String[] args) [0x005ef] in <f3fa022404f84f6f87ecbf3f76f088bf>:0 

Press any key to exit!
```

Attached I have the logfile while trying to start it in debug mode.

meee1, 2017-05-14:
can you verify is system.speech.dll exists?

meee1, 2017-05-15:
can you give the latest beta MP a try.
http://firmware.ap.ardupilot.org/Tools/MissionPlanner/beta/
I reverted to an older system.speech.dll

t2000, 2017-05-15:
The System.Speech.dll is is the same directory as the MissionPlanner.sh

With your reverted System.Speech.dll in the beta I am able to start MissionPlanner again. Thank you very much!

But nevertheless a commandline option would be nice in the case that you need a newer System.Speech.dll for windows and mono is not supporting it. So you would still be able to use MP but without speech then.

meee1, 2017-05-31:
the issue was actually around the fact that the newer system.speach was marked as a system library internally. so mono was skipping the load and looking within the mono system libraries. which it does not have a substitute for. so its actually a mono bug.

closed 2017-05-31 by meee1

branob, 2018-03-03:
Still having this issue with Mission Planner 1.3.52 on Linux Mint (Ubuntu). Mono installed with apt-get install mono-complete.

Any idea how to move pass this please?

$ ~/Downloads/MissionPlanner-1.3.52 $ mono MissionPlanner.exe
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35 

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
INFO Program - ******************* Logging Configured *******************
INFO Program - 64bit os True, 64bit process True
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883_OLD or 1 
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype DEVTYPE_AK8963 or 4 
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2 
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883 or 7 
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2 
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype DEVTYPE_AK8963 or 4 
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 22 or DEVTYPE_ACC_MPU9250 
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype 17 or DEVTYPE_ACC_LSM303D 
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 22 or DEVTYPE_ACC_MPU9250 
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 36 or DEVTYPE_GYR_MPU9250 
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 4 devtype 34 or DEVTYPE_GYR_L3GD20 
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 36 or DEVTYPE_GYR_MPU9250 
**FATAL Program - Fatal app exception
System.IO.FileNotFoundException: Could not load file or assembly 'System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies.**
File name: 'System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'
  at (wrapper remoting-invoke-with-check) MissionPlanner.MainV2:.ctor ()
  at MissionPlanner.Program.Main (System.String[] args) <0x4016d800 + 0x01caf> in <filename unknown>:0 
System.IO.FileNotFoundException: Could not load file or assembly 'System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35' or one of its dependencies.
File name: 'System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'
  at (wrapper remoting-invoke-with-check) MissionPlanner.MainV2:.ctor ()
  at MissionPlanner.Program.Main (System.String[] args) <0x4016d800 + 0x01caf> in <filename unknown>:0

meee1, 2018-03-03:
grab a copy of https://github.com/ArduPilot/MissionPlanner/raw/master/ExtLibs/System.Speech.dll

branob, 2018-03-03:
That did it, thank you. Now it starts but I'm getting bunch of other errors on console ...I'll try to tackle them first before I post back ...unless you'd like to see the errors?