ArduPilot/MissionPlanner #1554: "Load custom firmware" does not appear

MR444, 2017-05-14:
Cannot upload custom firmware. Buttons are gone in newest firmware.
#### Version
Newest
[image]

meee1, 2017-05-14:
goto MP settings.
 and change layout to advanced. 

then retry

closed 2017-05-14 by meee1

MR444, 2017-05-14:
Thanks for fast reply. Where can I find MP settings? I can not remember ever having changed this but I uploaded some beta firmwares last year. It is very confusing because the button appears first when checking for firmwares and is gone later:
[image]
[image]

Could you please add this information to manual? http://ardupilot.org/planner/docs/common-loading-firmware-onto-pixhawk.html

rmackay9, 2017-05-15:
It's Config/Tuning >> Planner >> Layout drop-down