ArduPilot/MissionPlanner #1555: Mission planner loading fmuv2 to Cube

hdtechk, 2017-05-22:
#### Issue details. 
_Please describe the problem, or desired feature_
Mission planner not recognizing pixhawk2.1 cube during update  it's loading fmuv2 instead of fmuv3 
#### Version
_What version was the issue encountered with_
3.5.0rc6
#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk2.1 cube fmuv3 
#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2017-05-31:
mp now loads v3 when it can.

closed 2017-05-31 by meee1

rmackay9, 2017-06-01:
great!!

hdtechk, 2017-06-01:
Thanks Michael