ArduPilot/MissionPlanner #1557: Arduplane: Failsafe message does not clear after event off

GrahamDyer, 2017-05-27:
#### Issue details
Received a battery failsafe today due to improperly set param, landed and the voltage did not go low again yet the failsafe message did not clear (and there is no way to clear/cancel it). 

The message "FAILSAFE" persists indefinitely even after landing and there is no further event re-occurance, only goes off after reboot. 

Throttle failsafe also does not clear after the event with the FAILSAFE message persisting indefinitely.

The message should perhaps persist for a reasonable amount of time after the event has cleared, say 30-60 seconds and then clear automatically, allowing for a repeat message if further events occur.

#### Version
MP v1.3.45

AP v3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Twin boom pusher

#### Hardware type
AUAV-X1 autopilot 

#### Logs
N/A

[image]

magicrub, 2017-05-28:
The battery library makes an incorrect assumption that only a single battery is used and it will always go down. It's a little more complicated than that so we need to add some more logic to handle cases like this.

meee1, 2017-05-31:
this is a plane code issue, MP is just displaying what it has been told. please raise and issue there

closed 2017-05-31 by meee1