ArduPilot/MissionPlanner #1558: Current displaying current 2  input PH2.1 RC7

madangler1, 2017-05-29:
#### Issue details
Current reading from Bat displaying current 2 input, both current and current 2 display same value, when using single power input on power 1 voltage displayed but no current, when using duel power modules Power 2 current displaying on both. 

Batt set to current pin 3 voltage pin 2 
Bat 2 set to current pin 14 , voltage pin 13. 

Flight log showing correct draw on Batt so believe it's MP issue

#### Version
Latest beta downloaded today 1.3.48 Build 1.1.6344.20357

#### Platform

[  ] Copter

#### Airframe type
Quad X 

#### Hardware type
Pixhawk 2.1 with Mauch 100a Sensor and Mini power brick as Power 2. 

#### Logs

meee1, 2017-05-30:
can you please provide a tlog?

madangler1, 2017-05-30:
Absolutely please see below. 

I have just created this log for a test to demonstrate.

https://www.dropbox.com/sh/bk8alfb12o7n7li/AAAykG4hRfDpEOQGDONon75Ua?dl=0

This is a PH2.1 with Mauch current sensor on Power 1 and the traditional brick on Power 2, it's configured correctly with Batt being set to look at pins 2 and 3 and Batt 2 is set to look at 13 and 14.

There is no current showing with any draw via Batt (Power1 ) input but voltage is correct, If I draw off power 2 then both Current and Current 2 show the same.   

If I disable Power 2 (Batt 2) sensing then Current shows nothing. 

After testing is appears regardless of settings Current is just mirroring Current 2 input and there is no way to display actual Current from Power 1. 

I have checked the flight logs and everything there is normal. 

You need anything else let me know, also in the folder is some logs from yesterday too when I was tuning that should be longer.

madangler1, 2017-05-30:
I should ad the log I created to test just now I did 2 things 

First is test with the motors running with draw through power 1 port only, MP displayed no reading on "Current" or "Current 2"  

Second is with motors off and draw through power 2 and 1 together, MP showed "Current"and "Current 2" draw correct. 

My power 2 port is a aux output that's wired after the sensor on power one so in normal circumstances Power 1 port should Show just on "Current" and Power 2 port Should show on both "Current"and "Current" 2.  

Just a final note when I swapped Power 1 and 2 around again the same issue where only input from Power 2 is displayed on both. 

Thanks and hope it makes sense.

madangler1, 2017-05-31:
I see you have fixed it, tested on latest beta today and it's resolved.

Thanks

meee1, 2017-05-31:
yep, should be fixed.

closed 2017-06-02 by meee1