ArduPilot/MissionPlanner #1561: NTF_DISPLAY_TYPE parameter missing in MP 1.3.48

iskymaster, 2017-05-31:
#### Issue details
Hi,

I want to use an external LCD display with Pixhawk 2.4.8 but it appears in MP there is no such parameter setting for NTF_DISPLAY_TYPE

#### Version
MP V 1.3.48

#### Platform
[  ] All
[  ] AntennaTracker
[ x  ] Copter
[ x ] Plane
[ x ] Rover
[  ] Sub

#### Airframe type
QUAD
#### Hardware type
PIXHAWK 2.4.8

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2017-05-31:
support for the OLED display is new for Copter-3.5 which is still in beta testing.
http://ardupilot.org/copter/docs/common-display-onboard.html

The beta is fairly stable but it's still got a few small issues we need to sort out.  If you're happy taking the risk of flying beta firmware, it can be loaded using the MP's install firmware screen.  On the bottom right there should be a "Beta Firmwares" link.  If you don't see it then it can be enabled by going to the Config/Tuning >> Planner screen and setting the Layout drop-down to "Advanced".

closed 2017-05-31 by rmackay9

iskymaster, 2017-05-31:
Thanks rmackay9 I will check it out.

Typically after how long a beta release goes GA or stable category?

rmackay9, 2017-06-01:
@iskymaster, it's hard to say when it goes stable exactly.  it depends upon how quickly we can get through any critical issues that are raised.  We've still got two RC input issues - they affect a small number of users but we're planning to resole these so we're at least a few weeks away from a stable release.

iskymaster, 2017-06-01:
Thank you kind of new here. By the way any advice on how to find and hire PixHawk related programmer? I am in need for one some custom development.