ArduPilot/MissionPlanner #1562: NTRIP Client doesn't work correctly, NMEA-GGA not periodically

Operator99, 2017-05-31:
#### Issue details
The NTRIP Client doesn't work correctly, if you want to use an offical service like SAPOS.

I asked them for the logfile entries and they told me, there is only ONE NMEA-GGA Message after the connection but the system needs periodically this NMEA-GGA Message to prevent a timeout.
Further the system needs the NMEA-GGA Message periodically for updating the RTK-Corrections.

So what's going on in the Mission Planner (NTRIP-Client) backround? Is there a refresh rate integrated for the NMEA-GGA message?

Hardware: ublox NEO M8P Rover-Version.

#### Version
Mission Planner 1.3.48  build 1.1.6330.31130

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad

#### Hardware type
Pixhawk

#### Logs

meee1, 2017-05-31:
currently no, I assume sapos is using a vrs system? or just using the nearest physical base?

Operatorr99, 2017-05-31:
You have the possibility to choose different Mount-Points with SAPOS. 

1. VRS -> Virtual Reference Station
2. MAC -> Master- Auxiliary Concept

The question is, what do I need for the Mission Planner and the NEO M8P Rover Module?
Which System? VRS or MAC ? Which messages? RTCM 1005, RTCM 1074, RTCM 1084 are enough?

meee1, 2017-06-02:
please try the latest beta

Ecky66, 2017-06-06:
Hi Operator99,

did you check the SAPOS Connection via the latest beta as meee1 suggested?
In the moment i have not a M8P to check :-(

Regards
Christian