ArduPilot/MissionPlanner #1566: Max. throttle is not acquired

sgrgpt, 2017-06-06:
#### Issue details
_Please describe the problem, or desired feature_
When I am pointing the stick of my radio controller in the top extreme position for max throttle, I am getting only about 20% of it.  
#### Version
_What version was the issue encountered with_
I am using APM 2.6 board with mission planner 1.3.48 with firmware version 3.2.1  
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter - THIS ONE**
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
'X' Quadcopter
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
APM 2.6
#### Logs
_Please provide a link to any relevant logs that show the issue_
Not really sure.

meee1, 2017-06-06:
can you confirm your transmitter end point and travels are set correctly?

sgrgpt, 2017-06-06:
the transmitter end points for throttle pin are 956 to 1901. and I don't well understand what you mean by 'travels'

magicrub, 2017-06-06:
How do you know you're only getting 20%? This sounds like a configuration error.

sgrgpt, 2017-06-06:
i was not able to get the escs callibrated through the software it showed an error- 'Set param error. Please ensure your version is AC3.3+' However my board is not able to download firmwares post 3.2.1. So i did manual calibration in which 1microsec signal corresponded to min throttle and 2 corresponds to max throttle. testing with a logic analyser a signal of only 1.2-1.3 micro sec was being obtained at max throttle.

narew, 2017-06-08:
In situation like this, first thing what I do is display value of ch3in (or another "channel in" which supports thrust) and ch1out, ch2out , ch3out ,ch4out  in "Quick" Window on FlightData page in MissionPlanner. Then connect wireless or by usb cable (without propeller on motors :) ) and check where is the problem: with input (RC transmiter) or with output (in autopilot config). 

You can also check:
- mixers on RC transmiter
- values of THR... or MOT... (in Config Tuning>Full Parameter Tree>THR..) if their setting occur in V3.2.1
- battery and battery failsafe voltage level.