ArduPilot/MissionPlanner #1568: Mission Planner 1.3.48 build 1.1.6330.31130 freeze while openning

ozzcet, 2017-06-08:
#### Issue details
OS win 10 
last update 2 days ago after that mission planner running. However yesterday suddenly when i try to open mission planner it freeze on splash screen. (not responding)
how is it fixed?

Best regards.

#### Version
Mission Planner 1.3.48 build 1.1.6330.31130 

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

[image]

meee1, 2017-06-08:
do you have an intel graphics card?

ozzcet, 2017-06-08:
[image]

ozzcet, 2017-06-09:
So? My friend how can i handle?  should i format my computer to setup again... i need to fix this to first autonomous fly, i am curious :-D

ozzcet, 2017-06-11:
Now it is working :) but i dont know why :D :D  Thanks.

amcmahon01, 2017-06-12:
I am having the same problem on Windows 7 (with Intel Graphics 530 and AMD Radeon R9 M375X) but have been unable to resolve it.  The latest version that works for me is 1.3.44.

amcmahon01, 2017-06-12:
Updating the AMD driver solved it.  I was previously running 15.30.1025 and have now updated to 17.10.1731.

ozzcet, 2017-06-17:
First time i get a chance to debug , and i get this ....

[image]

closed 2019-11-08 by ozzcet