ArduPilot/MissionPlanner #1569: Mission planner - Firmware 1.3.39.msi    -file corrpted- ?

Sas-LAB, 2017-06-09:
#### Issue details
Looking at the folowing link. 
http://firmware.ap.ardupilot.org/Tools/MissionPlanner/archive/

Tried to download the firmware version 1.3.39
http://firmware.ap.ardupilot.org/Tools/MissionPlanner/archive/MissionPlanner-1.3.39.msi

Try to execute this file at windows 10 I get an error. 

(This installation package could not be opened. Contact the application vendor to verify that this is a valid Windows installer package.) 

Is that normal?
I'm doing something wrong?
(just wanna install **this** version on the PixHawk FC) 

Thank you very much !
Sascha

#### Version
1.3.39 

#### Platform
[  ] Copter

meee1, 2017-06-09:
i just tried it and I got no issues.
please provide more info on the exact error

Sas-LAB, 2017-06-20:
Good afternoon,

Will try to record a video so you can see the exact error.
(sorry for the delay)

Sas-LAB, 2017-06-20:
**UPDATE**

Check the link today and **everything is working perfectly.**
(I could not reproduce the error)

Sorry for the troubles, didn't change anything, update....

Sas-LAB, 2017-06-20:
You can close the ISSUE

closed 2017-06-20 by meee1

PPL30, 2019-06-26:
почему не скачивается программа Mission planner - Firmware 1.3.39.msi