ArduPilot/MissionPlanner #1576: Unable to connect APM 2.8 to Mission Planner via bluetooth.

mukut03, 2017-06-13:
#### Issue details
Hello there, I am in the process of making my first Quadcopter but I am unable to connect my APM flight controller to mission planner using MAVlink. I am using a HC-06 bluetooth module and have set the baud rate to 57600. MP is running on Windows 7 64-bit. Selecting the correct COM port and baud rate in MP software, it tries to connect and it seems they have found each other (red led goes solid on the HC-06 and in top right corner of MP it shows connected) but then the MAVlink dialog box shows "trying to connect - timeout it 28s, 27s, 26s" and then just shows "connection failed". On clicking on details it shows "timeout" exception. Can you help me out please? I am almost about to give up on this project as it has been a month now and I cannot find any solution. I have searched extensively on the net and this site but non of them helped. I am a 9th grader and very passionate about robotics and automobiles. I would be very grateful to your help. Thank you.

#### Version
OS - Windows 7 64-bit; Mission Planner 1.3.48

#### Platform
Copter

#### Airframe type
Quadcopter

#### Hardware type
APM 2.8
#### Logs

[image]

magicrub, 2017-06-15:
Please post help requests to the forum at http://discuss.ardupilot.org. The Github issues list is for confirmed bugs and feature requests.

mukut03, 2017-06-15:
Thanks, this problem is also posted at ardupilot forum but can you please
elaborate on 'help requests' ? Thank you.

magicrub, 2017-06-15:
"help requests" mean things that can be handled by the thousands of volunteers who visit the forums and know enough about Ardupilot to help you through some common or popular problems that people have. You know, help. It does not mean grab the attention of a developer who is busy adding features or fixing bugs who will benefit millions of users so that he/she can tell you something that is found on a wiki.

Keep in mind, the APM 2.x hardware was depreciated many years ago due to it's extreme lack of resources. Please upgrade your hardware. If I could close this issue, I would.

mukut03, 2017-06-15:
Thank you for your quick reply. Can you suggest me any other hardware that
I upgrade my APM 2.8 to? I plan to try out the autonomous missions.Once
again thank you.

magicrub, 2017-06-15:
The Cube (aka pixhawk 2.1)