ArduPilot/MissionPlanner #1578: Mission Planner SITL -Copter

lvale, 2017-06-15:
I was asked how to use the simulator from within Mission Planner. From my understanding it starts a SITL via a sim_vehicle.py file. So far, so well. But contrary of what I'm used to run from a command line, this Mission Planner invoked SITL requires to setup the frame type for copter, keeps nagging about RC roll not configured and demands to have a 3D Accel calibration done. Short of moving the pc in all directions I really don't know how to instruct NON developer people to use SITL with MP to practice mission planning.

[image]
[image]

meee1, 2017-06-15:
broke via this commit
https://github.com/ArduPilot/ardupilot/commit/3a9ac2ef40129ec81ad6c537257574cf67f8b34f

meee1, 2017-06-15:
new beta pushed with fix

lvale, 2017-06-15:
thanks Michael

lvale, 2017-06-17:
Verified working on last beta

closed 2017-06-17 by lvale