ArduPilot/MissionPlanner #1579: Geo Ref Images (Cannot access disposed object)

tokm, 2017-06-16:
#### Issue details
Geo Ref Images:

Browse log, select either <log>.bin or <log>.log then click Pre-Process.  The output is displayed.  Click 'GeoTag Images' button.  Output shows 'GeoTagging...<image 1>' then nothing.  I think there is a thread doing the tagging that is trying to access a disposed object.  No error is thrown, it just sits there and the 'GeoTag Images' button becomes enabled again indicating some thread ended.  If I close and re-open the Geo Ref Images screen enough times I must be re initializing some object and I can eventually geotag the images.

#### Version
1.3.48 build 1.1.6330.31130

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A

#### Hardware type
N/A

meee1, 2017-06-16:
please post the exact error/screenshot

tokm, 2017-06-16:
2017-06-15 16:50:02,043 DEBUG MissionPlanner.Utilities.Tracking - v=1&tid=UA-43098846-1&cid=0510354a-bea7-4d85-9e39-5d47b3146d01&t=exception&an=Mission+Planner&av=1.3.48&cd=MissionPlanner.GeoRef.Georefimage&dp=MissionPlanner.GeoRef.Georefimage&exd=Cannot+access+a+disposed+object.%0d%0aObject+name%3a+%27TextBox%27.&exf=0&cd5=System.ObjectDisposedException%3a+Cannot+access+a+disposed+object.%0d%0aObject+name%3a+%27TextBox%27.%0d%0a+++at+System.Windows.Forms.Control.CreateHandle()&ul=en-US&sd=32-bits&sr=1920x1080&cd1=Microsoft+Windows+NT+6.2.9200.0&cd4=8&z=661732139 (:0) [38]
2017-06-15 16:50:02,056 ERROR Program - System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'TextBox'.
   at System.Windows.Forms.Control.CreateHandle()
   at System.Windows.Forms.TextBoxBase.CreateHandle()
   at System.Windows.Forms.TextBoxBase.SetSelectedTextInternal(String text, Boolean clearUndo)
   at System.Windows.Forms.TextBoxBase.set_SelectedText(String value)
   at System.Windows.Forms.TextBoxBase.AppendText(String text)
   at MissionPlanner.GeoRef.Georefimage.WriteCoordinatesToImage(String Filename, Double dLat, Double dLong, Double alt)
   at MissionPlanner.GeoRef.Georefimage.BUT_Geotagimages_Click(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnClick(EventArgs e)
   at MissionPlanner.Controls.MyButton.OnClick(EventArgs e)
   at System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   at MissionPlanner.Controls.MyButton.OnMouseUp(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam) (:0) [Base Thread]

meee1, 2017-06-16:
can you post part of the log before that? that is the error. but I need to know what was going on before you got to that point

tokm, 2017-06-16:
[MissionPlanner.log.2017-06-15.1.txt](https://github.com/ArduPilot/MissionPlanner/files/1080487/MissionPlanner.log.2017-06-15.1.txt)

meee1, 2017-06-16:
ok that logs shows you clicked the georefimage button more than once. and it did actualy tag the images. and then closed the screen before it had finished processing the second/third attempt

tokm, 2017-06-16:
No, it listed the first image in the output and sat idle as well as re-enabled the 'GeoTag Images' button as if it was done.

meee1, 2017-06-16:
the re-enable of the button was part of the bug. it always re eneabled even though it was still working