ArduPilot/MissionPlanner #1586: Image Tagging,  Cam Message Syncro not adding Omega, Phi, and Kappa to image exif data

LeonDwyer77, 2017-07-03:
#### Issue details
Cam messages are logging Omega, Phi, and Kappa to 2 decimal places.

CAM, 1330280308, 517348800, 1955, -36.8878398, 144.7969316, 301.84, 106.27, 301.04, 
**-0.28, 0.54, 3.78**

However, this data is not being added to the image metadata.  Having this data added to the exif information would greatly decrease processing time and geo-accuracy when making maps.

#### Version
MissionPlannerBet	Jun 21 01:10

#### Logs
https://drive.google.com/file/d/0BxZEUE3lH9TYSmg1SUJ4OFZwSE0/view?usp=sharing

meee1, 2017-07-03:
this is not possible. EXIF doesn't have options for this extra data. if you want to use this info you need to import a separate file with this information in. I know photoscan can do it, unsure on pix4d

LeonDwyer77, 2017-07-03:
pix4d can do it but this problem is that it uses the GPX stream not the CAM
messages - this adds further positioning errors.

I'll have to look at a way to somehow merge this data.

kikislater, 2017-07-03:
In standard exif paramters it's not available but in exif xmp metadata, you could store what you want.

meee1, 2017-07-04:
according to https://support.pix4d.com/hc/en-us/articles/202558539-Input-Files#labelA1_1
MP already generates a file that should be compatible. "locations.txt"

LeonDwyer77, 2017-07-04:
> Input-Files#labelA1_1

That's right it does. But.....
Pix4D uses the GPS log not the Cam Messages Log to tag images.
It is handy, it does a much better job at time offset geotagging than
Mission Planner or geo setter.

But that also means that positioning information is a bit random as
the  GPS log is out of step with the  Cam message synchro log, Cam message
synchro image tagging in Mission Planner is far superior in positioning
than all other options. except that it does not include  Omega, Phi, and
Kappa.

What I'm trying to achieve is Direct Georeferencing.  Which means I need
the GPS, Alt and Omega, Phi, and Kappa matched with the exact time the
image was taken.

Do you think It is posable to create a CSV file with images GPS and Omega,
Phi, and Kappa that Pix4d can import? I can provide you with a sample CSV
file the Pix4d can handle.

Otherwise, It looks more like this is a request for Pixd4 to implement a
cam message Syncro log reader/tager, and I dought I'll make much progress
there.