ArduPilot/MissionPlanner #1589: Question: Can I use Mission Planner to stream video and telemetry data to a back-end?

wickramagedsm, 2017-07-04:
I want to stream the video and telemetry data from the drone to a back-end control station (other than ground control) more like to a video management server and stream it from there to different clients.

Can I do this with the help of Mission Planner? Thanks.

meee1, 2017-07-06:
without know what transport you are using. i cant give you an answer.
i need more information about the system

wickramagedsm, 2017-07-10:
I was thinking of a TCP/IP based transport in a private network (but please recommend if there's a better way to achieve this). The requirement is, like how the video is streamed to the ground control station from the drone, we would like to stream the drone camera stream (and telemetry data) to a media streaming server which then will be streamed by clients from a central control station.