ArduPilot/MissionPlanner #1590: Batt Monitor broken in 1.3.48

iskess, 2017-07-08:
#### Issue details
After updating to 1.3.48 I get nonsense current readings. If I play the .tlog back on my computer with 1.3.46 it shows the proper readings. 

#### Issue details
Today I upgrade 2 of my planes to 3.8r6.  The following screen shots were taken on the bench at zero throttle. 

Before:
[image]

After:
[image]

Here is me pulling 38,300 Amps in flight. 
[image]

#### Version
1.3.48

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Sub

#### Hardware type
Pixhawk

#### Logs
https://www.dropbox.com/s/gty6b1331gm4v9t/2017-07-07%2015-37-51.tlog?dl=0

magicrub, 2017-07-08:
Looks like a units problem..

meee1, 2017-07-09:
can you try beta MP and see if its still an issue. I think this is to do with the new battery message scaling.

iskess, 2017-07-09:
I played back my tlogs from flights that displayed batt incorrectly, with the lastest beta it now displays as it should. Thanks for the quick response.

closed 2017-07-09 by iskess