ArduPilot/MissionPlanner #1592: Matlab export broken?

RoelHelsen, 2017-07-10:
#### Issue details
I have been using the matlab export feature from mission planner for more convenient data processing. For this I use octave ( the linux equivalent of matlab). This has been working fine until verion 1.3.45 of missionplanner. However, for the later versions of missionplanner, this still generates a .mat file, but I cannot open it anymore from octave. When I try to open it, it gives the following error : "_error: load: error uncompressing data element (data error from zlib)_".
After some research, I found that other people have had the issue before (http://savannah.gnu.org/bugs/?48018) and that it is most likely related to the way in which the files are created.
Can anyone help me with this?

#### Version
MissionPlanner V1.3.46 and newer

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover
[  ] Sub
[x] MissionPlanner

meee1, 2017-07-11:
please try the latest beta MP

RoelHelsen, 2017-07-14:
Dear Michael,

I just tried with the latest beta MP, and now I can open the .mat files in octave again.
So, Thank you very much!

It took a bit longer because the beta version crashes on start-up on my computer. Now I tried with a different computer and there it started and I could do the .mat generation. These .mat files I can read again with octave.

Kind regards,
Roel helsen

meee1, 2017-07-14:
on the crashing pc, make sure you have .net 4.6.2 installed.