ArduPilot/MissionPlanner #1595: Auto Analysis dosen´t works

Fortesque0, 2017-07-12:
#### Issue details
I use Auto Analysis feature but returns an error

#### Version
MP 1.348.3 Beta
#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Trade Heli

#### Hardware type
pixhawk

#### Logs
[image]

smiltee, 2017-07-20:
Can you provide an input file?

Fortesque0, 2017-07-20:
Hello, 
Yes, I will upload some files soon.
Thanks

Fortesque0, 2017-07-29:
Sorry for delay. I am on holidays. 
[Auto_log_bad.zip](https://github.com/ArduPilot/MissionPlanner/files/1184816/Auto_log_bad.zip)

meee1, 2017-07-30:
for some reason MOT_PWM_MIN doesn't exist in that log. and is causing it to fail

meee1, 2017-07-30:
Traceback (most recent call last):
  File "runner.py", line 6, in <module>
  File "LogAnalyzer.pyc", line 246, in main
  File "LogAnalyzer.pyc", line 91, in run
  File "C:\ProgramData\Mission Planner\LogAnalyzer\/tests\TestMotorBalance.py", line 44, in run
    min_throttle = logdata.parameters["MOT_PWM_MIN"] / (logdata.parameters["MOT_PWM_MAX"]-logdata.parameters["RC3_MIN"])/1000.0
KeyError: 'MOT_PWM_MIN'