ArduPilot/MissionPlanner #1600: Laser altimeter over water causes RTL

russdelaneyuk, 2017-07-17:
#### Issue details
When using a laser altimeter and the "Terrain" option in mission planner, We have found that if the drone transits over water (pond) the drone will enter RTL.
This is due to the reduced range of the Laser Alt over water, it appears to see the loss of laser range as a terrain error (RTL)

Is there a way or addition that can be made to effectively continue mission in the event of? Using Baro only?
Thanks in advance
Laser Alt used SF11c

#### Version
3.4.6

#### Platform
Copter
#### Airframe type
Hex

#### Hardware type
Pixhawk2.4.8

russdelaneyuk, 2017-07-19:
Just to clarify, the SF11c has a range of 120m over land, this reduces over water to 40m.
If I plan a survey over a field say at 60m as soon as (2secs) it passes over water, in this case a small pond within the field boundary. The drone enters RTL, in most cases the drone would be clear of the water in say 10/15 secs and regain the Laser range of 60m. 
So having a time frame in seconds you could select before the drone enters RTL, or for the drone to revert to Baro height until the laser range is regained?
The whole purpose for the laser alt, is for us to get more accurate mapping resolution over undulating terrain. 
Hope someone could advise?

gmorph, 2017-07-20:
Can you attach a log file of the flight please?

What do you have AVD_F_RCVRY set to?

Thanks, Grant.

russdelaneyuk, 2017-07-20:
Hi Grant
Thank you for the reply, I will download the log file today and check the setting.
Thanks, Russ