ArduPilot/MissionPlanner #1606: Board Orientation

Arevlis, 2017-07-19:
#### Issue details
_Please describe the problem, or desired feature_
Unable to change board orientation
#### Version
_What version was the issue encountered with_
1.3.48.3
#### Platform
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_
Board is oriented yaw180 on frame, but I cannot calibrate accelerometers to reflect that. I was able to change AHRS ORIENTATION in an older version of MISSION PLANNER, but I do not see that parameter anywhere in this version.

rmackay9, 2017-07-20:
It's probably that the advanced layout needs to be enabled.
Config Tuning >> Planner, then select "Layout" to "Advanced"

Arevlis, 2017-07-20:
Thank you for your quick response. And yes, problem solved. Thank you!

From: Randy Mackay 
Sent: Wednesday, July 19, 2017 8:31 PM
To: ArduPilot/MissionPlanner 
Cc: Arevlis ; Author 
Subject: Re: [ArduPilot/MissionPlanner] Board Orientation (#1606)

It's probably that the advanced layout needs to be enabled.
Config Tuning >> Planner, then select "Layout" to "Advanced"

closed 2017-07-20 by meee1