ArduPilot/MissionPlanner #1612: can't install MP due to same or higher version of .NET Framework 4 has already been installed on this computer

mkeyno, 2017-07-26:
#### Issue details
I've tried to install the last version of MP but as I already install .net ver 4 .5  I can't install .net4 which is asked by MP to installation

same or higher version of .NET Framework 4 has already been installed on this computer.

#### Version
MissionPlanner-latest.msi
#### Platform
[  ] All
 

#### Airframe type
trycopter

#### Hardware type
APM 1

meee1, 2017-07-26:
MP required .net 4.6.2

itayguy, 2017-07-29:
it cannot be installed with VS 2017

itayguy, 2017-07-29:
go into VS installation and add .net 4.6.2 componnents

closed 2017-07-29 by mkeyno

ChrisThompsonTLDR, 2017-09-04:
Was anyone able to get around this?  It appears that Windows 10 Anniversary Edition comes with 4.7 and claims that it is an in-place update to 4.6.2.

[.NET 4.6.2](https://www.microsoft.com/en-us/download/details.aspx?id=53345) doesn't even appear to be available for Windows 10.

meee1, 2017-10-19:
4.7 will work fine as well