ArduPilot/MissionPlanner #1613: Extended tuning does not enable

timvand, 2017-07-26:
#### Issue details
Extended tuning is NOT visible in the Configuration tab, while Layout is set to advance on Planner.

#### Version
All

#### Platform
[X] All

#### Airframe type
Multirotor and fixed-wing

#### Hardware type
PH1/PH2

#### Logs
https://prnt.sc/fzxbia

[image]

meee1, 2017-07-26:
what firmware are you connected to? plane copter etc

timvand, 2017-07-27:
Plane.

meee1, 2017-07-27:
copter has a page called extended tuning. 
https://github.com/ArduPilot/MissionPlanner/blob/master/GCSViews/SoftwareConfig.cs#L57

plane has a page called Basic tuning.
https://github.com/ArduPilot/MissionPlanner/blob/master/GCSViews/SoftwareConfig.cs#L62

this has not changed in multiple versions. so I'm not seeing a problem/issue at all.

timvand, 2017-07-27:
What would be your suggestion/recommendation to accomplish this configuration in Plane?
[image]

Specifically the portion related to configuring CH7 OPT to Camera Trigger, on Plane FW?

meee1, 2017-07-27:
that feature/option is not available on plane. never has been.

timvand, 2017-07-27:
Understood.  So that feature is not available on Plane and thus the option is not configurable, or is there some other method to achieve the same result?

meee1, 2017-07-27:
you can trigger the camera from the GCS (flightdata > right click map). but I'm not aware of a method to trigger via a transmitter option, other than having a servo directly connected to that rx channel (ie manual bypassing).

timvand, 2017-07-27:
So, not possible in plane and possible in copter?  How can that be?

micaheli, 2018-10-08:
I agree that this is a ridiculous limitation.  Not having the ability to assign functions to switches at all is absolutely assinine.   Especially when functions of airplanes and multirotors have a lot of overlap.

The fact that this has not been re-addressed since 2017 hurts my brain.