ArduPilot/MissionPlanner #1614: MP 1.3.49 doesn't recognize a Ph2.1 board for fw update

mtbsteve, 2017-07-26:
#### Issue details
I just installed 1.3.49 release and have the problem that I can't update my Pixhawk 2.1 anymore. It simply won't detect and recognize the board at the beginning of the update. 
Otherwise MP works well, I can connect through USB, read and write parameters, HUD and all displays work seamlessly. So it seems not to be an USB driver issue.

I then downgraded back to the 1.3.48 version and update worked as usual. 
I then re-updated to the latest release 1.3.49 and the detection of the Ph failed again. Since I frequently updated Arducopter during the AC beta testing with 1.3.48 and prior betas w/o problems I assume a defect in the current release of MP. 

Setup: Win32bit laptop with the latest Win10 upgrades installed.

#### Version
MP 1.3.49 official release 

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk 2.1 (cube)

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2017-07-26:
This certainly works. Please check http://discuss.ardupilot.org for forum help to get this working for you. Check your USB drivers.

mtbsteve, 2017-07-26:
I am afraid but It  doesn't work with my setup as described above. 
I certainly tried to reinstall the drivers with 1.3.49. White MP can connect to the Ph2 at runtime w/o problems, it doesn't recognize the board when I try to run a fw update. 
Again, with 1.3.48 everything works.

meee1, 2017-07-26:
make sure you have the correct comport selected. this is something that changed recently. if you have the correct comport selected you will not have any issues. I will be changing this.

bobotus, 2017-07-29:
I had a the issue with my Beta batch PH2.1 running Copter fw3.4.4, MP 1.3.49. I could connect with Mission planner but was unable to load new 3.5 firmware.

Here's the things that I tried and was eventually able to load the new fw:
-Reset the fw parameters to defaults
-Unplugged power from PH2.1 power brick
-Unplugged all cabes and connected just with the micro usb to a USB3 port
Not sure what did the trick for me, but I was eventually able to load the new fw. Hope this helps.

mtbsteve, 2017-08-20:
I need to add to this issue. 
In the meantime I tried the last stable as well as the latest beta of MP to update my Pixhawk 2.1.  
By no means the board is detectable. 
I tried as well on a brand new installed 64 bit Win 10 PC w/o success regardless if I select the com port before or not. 

MP 1.3.48 however  works perfectly well on all my PCs. 

I am using the "green cube" - can that be the reason why MP 1.3.49 cannot detect the board??
@meee1

proficnc, 2017-08-20:
@mtbsteve 
you must load fmuv3 code onto the cube.

also, if this is going into a solo, please load all the correct firmware onto the Solo first using Solex, before attempting to update your cube.

mtbsteve, 2017-08-20:
Philip I have the GC in my DIY drone, not Solo, mounted directly on the carrier board and connected via USB to the PC. 
MP 1.3.49 including the latest MP beta do not recognize the cube for update - I tested on 3 PCs now. 
However I can connect at runtime, so it's not a driver problem. 
MP 1.3.48 works fine for update though. 
I would appreciate if someone could test the update with a green cube in particular. 

Unless someone fixes this defect in MP 1.3.49, the only workaround is to keep a 1.3.48 version of MP on a PC for future fw updates.

proficnc, 2017-08-20:
Your 100% sure you are trying to load an fmuv3? Not FMUv2?

mtbsteve, 2017-08-21:
How do I distinguish between V2 and V3?
Here is what I do:
1. connect the cube to the PC
2.  select the com port (COM 27 on my PC - there is no distinction between fmuv3 or v3)
3. select the copter type icon in MP (Quad  with AC 3.5.2)
4. unplug and plug back in the board as prompted 
get the message after 30 sec that the board cannot be detected

meee1, 2017-08-21:
please try the latest beta MP

mtbsteve, 2017-08-21:
@meee1 
Michael I tried with the latest MP beta today. 
Pixhawk 2.1 green cube: no success. Board won't be recognized as before 
Pixhawk 2 (old Solo cube) update works instantly. Board is correctly recognized, and v3 arducopter fw is automatically installed. 

I seriously believe that there is a defect in MP 1.3.49 onwards which prevents the recognition of a Pixhawk 2.1 green cube. (again, 1.3.48 works)
[image]

meee1, 2017-08-21:
please upload c:\programdata\missionplanner\missionplanner.log

meee1, 2017-08-21:
the code used to upload to a 2 and a 2.1 is exactly the same. so I'm not sure how it can be failing. hence the log request

mtbsteve, 2017-08-22:
@meee1 here is the link to the latest log. It should include several failed attempts with the GC and one successful update with the Ph2 cube.
https://www.dropbox.com/s/zsy4oyz06l7y210/Mission%20Planner.zip?dl=0

jfitter, 2017-09-01:
I can confirm this issue. Two computers.
Win7 desktop, I7, GT680, etc.
Win7 Dell laptop, I7, etc.
PixHawk 2.4.8
3D Robotics drivers 2.0.0.9

Both exhibit the same behaviour.
MP 1.3.49 cannot recognize the board and thereafter cannot upload firmware. Various messages are thrown. Cannot detect board, No heartbeat detected, Firmware update failed. This version does work correctly with APM2.5 board.
Reverting to MP 1.3.48 results in correct behaviour for the PixHawk. 
MP 1.3.45 also exhibits correct behaviour.

ns-1m, 2017-09-27:
I have the same problem in the last 2 days with Pixhawk 1 ver 2.4.7. MP 1.3.44 works before. Now, with MP 1.3.49, everything goes kaputt. Unplug and plug USB mayheim. I used 5 USB, same results.

geofrancis, 2017-09-27:
I have always had issues with flashing flight controllers using mission planner, I always keep a copy of apm planner2 installed just for updating flight controllers.

laurenweinstein1, 2017-11-27:
Same problem here. Failure to detect board on all versions past .48 -- including 50.9.

syediddi, 2020-06-08:
I have the same problem, bought a original pixhawk board from 3dr and it won't flash in the newer version of Mission Planner, however as @mtbsteve said, I installed version 1.3.48 and it worked. I tried in 3 different computers and same result, it says unable to detect the board