ArduPilot/MissionPlanner #1617: MP shows gps raw location but not ahrs location

itayguy, 2017-07-28:
#### Issue details
the symbol shown on the map uses gps raw data , but not estimated data 
whichis very jumpy.

#### Version
all
#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type

#### Hardware type
pixhawl v2
#### Logs

magicrub, 2017-07-28:
which symbol?

itayguy, 2017-07-28:
MAV

itayguy, 2017-07-28:
bytearray = MAV.packets[(byte)MAVLink.MAVLINK_MSG_ID.GPS_RAW_INT];
                    if (bytearray != null)
                    {
                        var gps = bytearray.ByteArrayToStructure<MAVLink.mavlink_gps_raw_int_t>(6);

                        if (!useLocation)
                        {
                            lat = gps.lat * 1.0e-7;
                            lng = gps.lon * 1.0e-7;

itayguy, 2017-07-28:
I would expect it to show this:
    mavlink_msg_global_position_int_send(
        chan,
        fix_time,
        current_loc.lat,                // in 1E7 degrees
        current_loc.lng,                // in 1E7 degrees
        (ahrs.get_home().alt + current_loc.alt) * 10UL,      // millimeters above sea level
        current_loc.alt * 10,           // millimeters above ground

meee1, 2017-07-28:
@itayguy did you check what "useLocation" is for?

itayguy, 2017-07-29:
so if I get "mavlink_global_position_int_t" I use it , and if not I use "mavlink_gps_raw_int_t" ?

meee1, 2017-07-29:
I don't know what your trying to archive. so cant comment.
to answer the original issue listed here. I don't believe its an issue at all. because its already using the best location available at the time.

closed 2017-07-29 by meee1