ArduPilot/MissionPlanner #1618: "VSync" Error in Design Time

jvasanthan, 2017-07-29:
#### Issue details
Hi,
I have pulled the latest Mission Planner Source code and built it using Visual Studio 2017 Community Edition. I added a button in the FlightData.Designer and tried to save the file, it threw up an error window as shown below. I found that when I cleared the ContextMenuStrip Option for the "hud1" component then I am able to save it.
I tried it in Windows 7 & 10 and i see the same error window while saving the FlightData.Designer. I read in the forum that it is related to openGL but not much of help. 

Could anybody please point me in right direction. 

Thank You

James

[image]

#### Version
Source code pulled on 27th July 2017

#### Platform
[ ~ ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad

#### Hardware type
Pixhawk

#### Logs
-

grvgoel1, 2017-10-05:
Is your problem solved? I am also facing the same problem.

jvasanthan, 2017-10-05:
I tried the following:
1. Disabled the context menu of hud and I didn’t get the vstnc error.
2.When I make change and do save(not save all) I don’t get the vsync error

Let me know if it helps

grvgoel1, 2017-10-05:
okay i will try it and get back to you on that

grvgoel1, 2017-10-06:
So I tried your solution but it did not work for me. Here's what I did, I set the bool of isShowContext to false. I hope that was what you meant. If not, then could you be a little more specific about which code to alter? Thanks!

jvasanthan, 2017-10-06:
pls disable the ContextMenuStrip of hud

grvgoel1, 2017-10-07:
@jvasanthan I cannot find the option to disable the contextmenustrip for hud, can you please pinpoint the file name or the code? Thank you!

meee1, 2017-10-09:
please update your source code, as i believe this is resolved in newer versions

closed 2017-10-09 by meee1