ArduPilot/MissionPlanner #1619: Mission planner Gstreamer Pipeline setup

roque-canales, 2017-07-29:
I use latest mission planner 1.3.49 + latest GSTREAMER msi.

I have no issues for open the pipeline below through windows cmd: 
gst-launch-1.0 rtspsrc location=rtspu://admin:admin@192.168.100.103:554/cam1/h264 latency=50 ! decodebin3 ! autovideosink

Quality and lag is perfect.

But I can't open it though Mission Planner > vidéo set gstreamer source > ctrl+F Gstream.

Does anybody could help and tell me how to?

meee1, 2017-07-29:
right click the hud > video> set gstream source and try

gst-launch-1.0 rtspsrc location=rtspu://admin:admin@192.168.100.103:554/cam1/h264 latency=50 ! decodebin3 ! avenc_mjpeg

roque-canales, 2017-07-31:
Thank you, but this no work....

roque-canales, 2017-07-31:
Hello Michael,

Thank you for your help,

I tried what you told to me, but this no work.

From where this issue could come?

Thanks you so much for your help.

meee1, 2017-07-31:
Try creating a pipeline that doesn't use decodebin

meee1, 2017-08-01:
what version of gstreamer did you install.

ive tested with 1.9.2

meee1, 2017-08-01:
does 
gst-launch-1.0.exe videotestsrc pattern=ball ! queue ! x264enc ! rtph264pay ! queue ! decodebin ! autovideosink
work?

roque-canales, 2017-08-01:
Yes!

gshmakov, 2017-08-08:
Confirm video freeze in mission planner with h264 rtp payload. When I run same pipeline through cmd all works well.