ArduPilot/MissionPlanner #1620: Mission Planner incorrectly reports that there is a new firmware version available when there is not.

george-hawkins, 2017-07-29:
#### Issue details
Mission Planner incorrectly reports that there is a new firmware version available when there is not. See the attached screenshot - Mission Planner reports that APM:Copter V3.5.0 is available but the latest stable version number shown in <http://firmware.eu.ardupilot.org/Copter/stable/PX4-quad/git-version.txt> is V3.4.6, which is the version I have installed already on my Pixhawk (this can also be seen in the screenshot).

#### Version
1.3.49

#### Platform

[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
X quadcopter.

#### Hardware type
Pixhawk

#### Screenshot
[image]

meee1, 2017-07-29:
its not a bug. your looking at the wrong url. and 3.5 is out.

closed 2017-07-29 by meee1

george-hawkins, 2017-07-30:
Apologies - I see now from looking at the MP source that I should have been looking at [`ArduPilot/binary/.../Firmware/firmware2.xml`](https://github.com/ArduPilot/binary/blob/master/Firmware/firmware2.xml) and worked from there to [`stable/PX4`](http://firmware.eu.ardupilot.org/Copter/stable/PX4/) rather than [`stable/PX4-quad`](http://firmware.eu.ardupilot.org/Copter/stable/PX4-quad/) as I'd assumed.

But what do you mean by "the issue is more around the realy old version of MP in that screenshot"? As you can see in the title bar it says **1.3.49** and according to [ChangeLog.txt](https://github.com/ArduPilot/MissionPlanner/blob/master/ChangeLog.txt) 1.3.49 is the latest version (released 20-7-2017). It's the version I downloaded from the [Mission Planner installation page](http://ardupilot.org/planner/docs/common-install-mission-planner.html) a few weeks ago.

[image]

meee1, 2017-07-30:
@george-hawkins  sorry misread the version number.