ArduPilot/MissionPlanner #1628: Geo ref images No AMSL in CAM Messages Error

aloerch, 2017-08-07:
#### Issue details
Using Here+ RTK base and rover, and camera triggering by waypoint. Using the Geo Ref tool with "CAM Message Synchro" enabled and with "Use CAM Messages" checked/unchecked (I tried both) and "Use AMSL Alt" checked, I get the following report when I click pre-process:

```
Using AMSL Altitude True
Reading log for GPS Messages in order to get AMSL Altitude
Log Read for GPS Messages
Log locations : 4233
Reading log for CAM Messages
Log Read with - 98 - CAM Messages found
Read images
Images read : 98
Photo DSC04895 processed from CAM Msg with 0 ms shutter lag. AMSL Alt NOT found
Photo DSC04896 processed from CAM Msg with 0 ms shutter lag. AMSL Alt NOT found
Photo DSC04897 processed from CAM Msg with 0 ms shutter lag. AMSL Alt NOT found
Photo DSC04898 processed from CAM Msg with 0 ms shutter lag. AMSL Alt NOT found
Photo DSC04899 processed from CAM Msg with 0 ms shutter lag. AMSL Alt NOT found
```

and so on....

However, the CAM messages in the log do show both an "Alt" and a "GPSAlt" column, both of which appear to be reporting AMSL. I'll attach a screenshot of that. 

If I use the "time offset" option instead of the CAM option, things start to get pretty wonky. For example, using "time offset" with AMSL checked results in my images being geotagged with AMSL, but with one big problem: Some images (about 10%) are being geotagged with an AMSL that is very close to the RTK base AMSL, which would mean that my drone would have dropped out of the sky, hit the ground, and immediately climbed back up (if this logging by the Geo Ref tool was accurate). I've attached the graph of the GPS Alt for the flight, to show that my UAS maintained a constant altitude until it was time to land (no 100m drops in the middle of the flight). I doubt very much that the "time offset" would be a great choice for me, given this problem. Besides, the CAM messages are exactly what I need, and they have the data I need... it's just that the Geo Ref tool doesn't seem to be looking in the right place for the AMSL for CAM messages. 

#### Version

1.3.49.1 Build 1.1.6417.13138 (I tried the beta just in case it had already resolved this issue, but it had not)

AND

1.3.49 Build 1.1.6410.20232

#### Platform
[ # ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
octa

#### Hardware type
pixhawk 2.1 w/ edison installed
[image]
[image]

aloerch, 2017-08-09:
While I believe this issue is a bug and not the intended behavior of the Mission Planner's Geo ref images tool, I have figured out a workaround. The workaround does not geotag the images, however, instead it requires using a separate text/csv file for image locations:

1. Use the geo ref tool to load your log and photos.
2. Select the Cam Message Synchro and check the box use cam messages
3. Uncheck the Use AMSL box
4. Run the pre-process step, and then click the locations button.
5. Now, in your photos directory you will have several "locations" file, I use the locations.csv file. The altitude listed in this file will not be what you want, if you want AMSL. Open this file in an application like MS Excel (or create a script to do all of this automatically if you're ambitions)
6. Open your pixhawk log in Excel too, and filter out all message types except the "CAM"  messages
7. If you used an RTK system on the drone, you probably want the GPSAlt column, and if you did not use an RTK system you might prefer the Alt column. Select either GPSAlt or Alt, and copy the column.
8. Go back to your locations.csv file and replace the Alt column with the copied GPSAlt/Alt column from the pixhawk log's CAM messages.

Now, you have a locations.csv file that has X,Y, and Z from the CAM messages, and the associated image names. In something like Agisoft Photoscan, you can use this csv instead of the image EXIF data. 

Of course, it would be much simpler and less error prone for Geo ref images tool to function properly by allowing the selection of AMSL for CAM messages.

I've noticed another issue posted here where people have asked for the option of choosing GPSAlt over Alt for RTK-based drone flights. I don't see this as the same issue, since AMSL is not being found in the cam messages by the tool at all, even though AMSL is there. Of course, I agree with the other posted issue... it should be possible to select whether one wants to use baro Alt or rtk (gps) Alt for their image tagging.

aloerch, 2017-08-12:
Please let me know if there's anything else I can do to contribute to fixing this bug in the georef tool.

geofrancis, 2017-09-30:
any progress on this?

geofrancis, 2017-10-05:
there is an interesting comment on line 1104 on https://github.com/ArduPilot/MissionPlanner/blob/master/GeoRef/georefimage.cs

            // If we are required to use AMSL then GPS messages should be used until CAM messages includes AMSL in the coming AC versions
            // Or if the user enter shutter lag and thus we have to look for GPS messages ahead in time

and the error message is at line 1180
logAltMsg = "AMSL Alt NOT found";

geofrancis, 2017-11-05:
from what i can tell the problem is here 
`
                    if (useAMSLAlt)
                    {
                        cameraLocationFromGPSMsg = LookForLocation(p.Time, vehicleLocations);
                        if (cameraLocationFromGPSMsg != null)
                        {
                            logAltMsg = "AMSL Alt " + (cameraLocationFromGPSMsg.Time - p.Time).Milliseconds + " ms away" +
                                        " offset: " + (p.ShotTimeReportedByCamera - dCAMMsgTime).TotalSeconds;
                            p.AltAMSL = cameraLocationFromGPSMsg.AltAMSL;
                        }
                        else
                            logAltMsg = "AMSL Alt NOT found";
}`

but im not sure how to fix that to get GPS ALT.

meee1, 2017-11-05:
can you provide some sample data? as I don't see the actual issue
the log would help as well

geofrancis, 2017-11-05:
its the same error as here #1032 if you use RTK with Hotshoe and try to geotag with AMSL alt using CAM messages you will get the error  "NO AMSL ALT FOUND"

meee1, 2017-11-05:
the message itself doesn't do anything, and doesn't stop the process

meee1, 2017-11-05:
can someone post a log. so I can see what we are looking at

aloerch, 2017-11-05:
I'll have to go back 3 months to when I first posted about this bug, to find some sample data and a log.  I'll try to post that  data in the next 12 hours. I'd honestly given up on the tool, choosing instead the arduous workaround solution I posted 3 months ago...

meee1, 2017-11-05:
from what I can tell the issue is only when using RTK. and is because the CAM message is not logging the rtk altitude.

I'm just trying to fully understand the problem.

aloerch, 2017-11-05:
I am glad this is getting looked at, and I do appreciate it. I would've had more available time too try and resolve the issue or provide additional feedback earlier; I'm a busy PhD student. When I'm home this evening, I'll have a look through my archived projects to find the logs and images that I referenced when i opened the GitHub issue. 

I did include in the original post a screenshot of the CAM messages in Mission Planner, and they do contain the RelAlt, Alt, and GPSAlt messages and values.

It's also really important to understand vertical measurements, especially in the context of whether someone should geotag their photos with the Altitude, GPS Altitude, AMSL, etc. I'm going to preface the next paragraph with the following info: My PhD is in Geography, where I've done a substantial amount of (and am an expert in) remote sensing and land surveying. When our drones are flying around logging their GPS locations, they're logging them in ITRF14 (equivalent of the current WGS84) for the X,Y (lat/lon) positions. For vertical (height/altitude) the positions are being logged as Height Above Ellipsoid (HAE) (not AMSL). To be able to actually tag an image with an AMSL, there needs to be a calculation of the HAE value to some Geoid model (which would give you actual AMSL). So, unless Mission Planner or the georef tool is performing a transformation of the altitude value from HAE to AMSL using a Geoid, there would never be an actual AMSL value to geotag the images with. Furthermore, the difference between an HAE and AMSL value is typically between 25-40 meters vertical.... When a someone wants to use their drone imagery for precision/accurate mapping, they will need to convert their altitude values from HAE to AMSL (or better, orthometric), unless the intent is to keep and report everything as HAE. So, what does this mean for what I'm saying about the georef tool? The GPS Altitude is HAE, and that's really the best we can hope for when tagging our images directly from the device logs. My recommendation is to ensure the georef tool can read the GPS Alt from the CAM messages and use that for geotagging the images. People who are interested in AMSL can use something like the NGS Vdatum application, or ESRI ArcGIS to do the vertical datum transformations.

aloerch, 2017-11-06:
Ok, looking at lines 1104 to 1180 here: https://github.com/ArduPilot/MissionPlanner/blob/master/GeoRef/georefimage.cs

I *think* I've come to the issue. @meee1 maybe you can tell me if I'm correct...

At the moment, Georef only tags images based on either AMSL or RelAlt, where AMSL is the height above some geoid, which has been implemented elsewhere in ArduCopter, and which lets a user define during flight planning and know after the fact a drone's height above mean sea level. CAM messages do not currently include the AMSL calculation to get from GPS Alt to AMSL, meaning that one cannot Georef their images based on CAM messages with anything other than RelAlt within the Georef tool. Is that correct so far?

So, the use of RTK also implies an importance to the reliance on the CAM messages for geotagging the images, specifically because the CAM messages should contain the coordinates (x,y,z) of where an image was taken, without performing a time comparison between image time and GPS time. We use CAM messages to know exactly where an image was taken based on the presence of the image (and order in which the images were taken) NOT based upon when an image was taken (+/- however many milliseconds). 

Is it correct then, that geotagging the images with CAM messages would work if AMSL were added to the CAM messages? Given the time that's passed without that being done, would it be simpler to make "Use GPS Alt" an option? An argument for "Use GPS Alt" would definitely be that people (such as myself) who are doing precision & accurate work would likely want to choose which Geoid model and vertical datum to use, rather than having MP/AC choose a geoid (I'm not sure even which one is used currently to get at AMSL in ArduCopter) for us.

aloerch, 2017-11-06:
Ah, it looks like Mission Planner and ArduCopter use EGM96 (the geoid most recently created for calculating MSL globally from the WGS84 ellipsoid). If you live in a country/continent that doesn't have a more accurate local geoid, then this is a good option. If you live someplace else (Europe, North/South America) and you are doing accurate mapping of absolute positions, then you would want to use a more localized Geoid for getting the AMSL values of your images. For example, converting the GPS Altitude (HAE) in the United States using EGM96 will be less accurate than converting your GPS Altitude using GEOID12B. While I don't think it's necessary for the georef tool provide multiple choices of geoids, I DO think it's a best practice to make a GPS Altitude option available for geotagging. 

People who are using RTK/PPK systems probably don't want an AMSL from EMG96 (whether they know that or not). If you were to add the Use GPS Alt option in place of (or addition to) fixing the AMSL issue, that would be the best solution.

norimboo, 2019-03-04:
Hi I am getting the same error "AMSL Altitude Not found" Please see my log:
[(http://fotostacja.pl/log.zip)] For now I am using Here 2 GPS but I wiuld like to use Here2 + Here+ combo to get RTK position