ArduPilot/MissionPlanner #1629: Feature request: easy selection of the internal/external sensor, like compass and barometers

robustini, 2017-08-10:
#### Issue details
What unfortunately is missing on the Planner is a clear section where the system shows how many peripherals are recognized from the flight control when booting, such as compasss and barometers.
For example in my S900 with Pixhawk2 i've in total 7 compass (five external) and three barometers (one external).
The Planner should clearly show which sensor was found, as it was assigned by the system by default, and what is its current order by numbering it.
Example:

"Found 7 compass:

0 - External 1 (I2C)
1 - External 2 (I2C)
2 - External 3 (I2C)
4 - External 4 (CAN BUS)

etc. etc...

And through a drop-down menu you can choose which ones to use from the system, also showing the current default.
This can be done with the "full parameters list", but for the average user is somewhat silly.

#### Version
The latest version of Mission Planner

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
It's not important

#### Logs
It's not important

proficnc, 2017-08-10:
Yes, I really would like a graphic that showed them all and allowed you to assign primary secondary etc!

robustini, 2017-08-10:
Yeah, and not just for the compass, but also for barometers and gps.

CharlieRome, 2017-08-10:
I agree, this would be very useful!

CharlieRome, 2017-08-10:
....maybe even with a "Test device" button next to each one... :)