ArduPilot/MissionPlanner #1636: "Change Speed" button doesn't work with ArduPlane 3.8

Bill-B, 2017-08-29:
#### Issue details
The 'change speed' button on the 'Actions' tab doesn't work with ArduPlane 3.8.0.

Looking into GCSViews/FlightData.cs, could it be because the ARSPD_ENABLE param was changed to ARSPD_TYPE for 3.8?
 
See https://github.com/ArduPilot/ardupilot/commit/14a1f559c70fcc08d90dc783a23175cf63483349

(I've only tested this in plane in SITL)

#### Version
MP 1.3.49, ArduPlane 3.8

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Plane (SITL)

#### Hardware type

#### Logs

mechbox, 2017-12-12:
I have exactly the same issue. I´ve just realised that Do_change_speed mavlink order actually works however the button still not working. 

Any solution?

Bill-B, 2017-12-21:
I'm not sure how the Do_change_speed mavlink command works, but the Change speed button just changes the value of the param "TRIM_ARSPD_CM". The issue with the Change speed button is due to some checks that are performed when the button is pressed. Flightdata.cs is looking for the value of the param "ARSPD_ENABLE", however that was changed to "ARSPD_TYPE" in Plane 3.8.

custeve, 2018-05-16:
Very major issue for us. Any fix?????

Bill-B, 2018-08-09:
Fixed as of MP version 1.3.57. See https://github.com/ArduPilot/MissionPlanner/commit/ee5e1f83464525c0769d2dfa4e89d254674d266e

closed 2018-08-09 by Bill-B