ArduPilot/MissionPlanner #1638: Make advanced layout the default

tridge, 2017-09-05:
I'd really like the full parameter tree to be available for all users. I've seen users (including Leonard!) spend a lot time being frustrated as they can't work out how to display the full parameter list as the "advanced view" option isn't really obvious (both not obvious that its needed, and where to find it)
Could we default to advanced layout at least on plane and heli? We discussed this on the weekly dev call too, and everyone on the call wanted advanced to be default for all vehicles.
Thanks!

rmackay9, 2017-09-05:
I agree.  Personally I prefer the full-parameter-tree to the full-parameter-list in case that makes a difference.

meee1, 2017-10-09:
done

closed 2017-10-09 by meee1