ArduPilot/MissionPlanner #1640: HW Flow control not supported

amilcarlucas, 2017-09-13:
#### Issue details
I have setup my 3DR radios like this:

* remote radio:
** RTS CTS activated in the Sik radio configuration
** BRD_SER1_RTSCTS parameter set to 1 (enable)
* Local radio
** RTS CTS deactivated in the Sik radio configuration

This configuration works fine with Mission Planner.

Now I activate RTS CTS on the Local radio in the Sik radio configuration, and restart both mission planner and the autopilot.
# Mission Planner times out and complains that it can not receive heartbet packets
# Close Mission Planner
# Open MavProxy
# MAVproxy works fine
# Close MavProxy
# Open Mission Planner
# Mission Planner times out and complains that it canot receive heartbet packets

#### Version
1.3.49

#### Platform
[  ] All
[  ] AntennaTracker
[ *] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
OctaQuad

#### Hardware type
pixhawk, official 3DR radios

meee1, 2017-09-13:
can you please try disabling reset apm on connectin in the MP options. this should not touch the rts settings

amilcarlucas, 2017-09-14:
I disabled "Reset APM on USB connect" in the MP options ... It still does not connect.

The RTS CTS radio setings do not get changed. The only issue is that MP can not directly connect (via USB) to a 3DR radio that has RTS CTS set.

amilcarlucas, 2017-09-14:
MP can however still configure the Sik Radios. That does work, regardless of RTS CTS.
[image]

amilcarlucas, 2017-09-27:
Tested in Qgrouncontrol. There it works as well.

meee1, 2017-09-27:
there is a fix in MP beta

closed 2017-10-09 by meee1

amilcarlucas, 2017-10-09:
Tested in 1.3.50. It works fine.

Thank you for fixing this.