ArduPilot/MissionPlanner #1641: OpenFileDialog of "Review a log" feature should remember last location path

SergeyBokhantsev, 2017-09-15:
#### Issue details
When clicking on "Review a log" button (DataFlashLogs tab) then the FileDialog always in the C:\Users\user\Documents\Mission Planner\logs folder.

It will be more sense to remember the last folder instead. 
For example, if I had previously opened a log from my flash drive (d:/Logs), then clicking [Review a log] once more, the FileDialog must open that folder on my flash drive. Not the default Mission Planner\logs folder.

This looks minor but it is very annoying though :)

#### Version
MP 1.3.49

meee1, 2017-09-17:
this behavior is by design.

you can change the default log location though to your d drive.

SergeyBokhantsev, 2017-09-20:
@meee1 I do understand the design that **initial** folder (i.e. when we started MP) is a default_log_location.
But if I opened a log from some another location, it is very conveniently if MP will remember that location until next restart.

Is there a reason to not to behave like that? 
Many (or most?) of the applications does remember last used folder, so this really doesn't look like a design but like a defect in MP.

SergeyBokhantsev, 2017-09-28:
@meee1 could you please look at this small PR?
https://github.com/ArduPilot/MissionPlanner/pull/1653

closed 2017-10-09 by meee1