ArduPilot/MissionPlanner #1643: Flight Planner Altitude "Type" box disappearing

jagrandle, 2017-09-18:
#### Issue details
_Please describe the problem, or desired feature_
After starting any of the above versions of Mission Planner I go to the "Flight Plan" tab. Even though I have not made a flight plan yet, there is the green Waypoint bar at the lower edge of the screen and there are certain headings for WP Radisu  Loiter Raidus, Default Alt then the drop down box for selecting whether the altitude is:
a) Relative
b) Absolute
c) Terrain

As soon as I connect the Solo and go back to the Flight Plan page this drop down box is gone !!
This means there is not way to verify or change what type of altitude is being used for a mission.

I have attached a screen shot showing the drop down box that I mean for the sake of clarity.

It does not matter whether I have loaded a flight plan or not, and nothing seems to bring it back short of disconnecting and re-starting Mission Planner.

Thanks,
Jag

#### Version
1.3.38 and 1.3.43 and 1.3.49

#### Platform
[ ] All
[  ] AntennaTracker
[X  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
3DR Solo

#### Logs

[image]
[image]

jagrandle, 2017-09-25:
Bump ... !!

Sorry to do this but we have a trial coming up and we would like to use the "Terrain" height for the first time on it. It would be REALLY comforting to be able to confirm what type of height is set before we fly !!

Thanks

meee1, 2017-09-25:
the solo doesn't support those frameworks.

run opensolo, and it will work.

closed 2017-10-09 by meee1