ArduPilot/MissionPlanner #1644: Mission Planning

lvale, 2017-09-19:
When generating some of the default surveys, (easiest example is circle with default values) MP creates too many items for the mission. Should MP planning tools cap the number of mission items to the max that ArduPilot vehicles support ?

meee1, 2017-10-09:
i never put a limit on it because moving forward aka linux, i didnt want to create an artificial limitation. so it has always been open ended. at this point in time i dont really plan on changing this. unless i can be convinced otherwise.

closed 2017-10-09 by meee1

lvale, 2017-10-09:
@meee1 I had a PR to raise the 724 WP on missions for the Linux boards, but it has never been merged, so in reality Linux boards have the same limits as the STM based boards.
I might resubmit the PR again, because now I have a Navio2 board to test. I've raised the limit to 7000 WP.
But currently if a user generates a mission, 95%+ will have to manually edit the generated mission file.