ArduPilot/MissionPlanner #1647: Support requested for graphing FFT from ISB messages in DataFlash files

peterbarker, 2017-09-22:
#### Issue details

We have a new feature coming in ArduPilot which allows fft-able data to be contained in many dataflash files, rather than just those with raw IMU sampling data enabled.

There is a new format field type, `a`.  It is an array of 32 16-bit signed integers.

There are two new messages, `ISBH` and `ISBD`.  One contains header information for a batch sample, the other the data for the batch sample.  Many `ISBD`s are expected for a single `ISBH` message, and one is related to the other using the `N` field.

The `ISBH` message contains `type` and `instance` fields, with `type` specifying an IMU sensor type (gyro or accel) and `instance` the instance number of that sensor on the flight controller.

#### Version

These patches are as yet unmerged into master.  https://github.com/peterbarker/ardupilot/tree/fft is the WIP for support.  We do not expect the format of the messages to change before that's merged in.

#### Platform
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Logs

[df-isb-fft-sample.bin.zip](https://github.com/ArduPilot/MissionPlanner/files/1323141/df-isb-fft-sample.bin.zip)

```
pbarker@bluebottle:~$ mavlogdump.py df-isb-fft-sample.bin --t FMT | grep ISB
Skipped 443 bad bytes in log at offset 3827, type=(163, 149, 248)
2017-09-22 09:33:08.14: FMT {Type : 248, Length : 29, Name : ISBH, Format : QHBBHQf, Columns : TimeUS,N,type,instance,mul,SampleUS,smp_rate}
2017-09-22 09:33:08.14: FMT {Type : 249, Length : 207, Name : ISBD, Format : QHHaaa, Columns : TimeUS,N,seqno,x,y,z}
```

```
pbarker@bluebottle:~$ mavlogdump.py df-isb-fft-sample.bin --t ISBH,ISBD | head
Skipped 443 bad bytes in log at offset 3827, type=(163, 149, 248)
2017-09-22 09:33:08.14: ISBH {TimeUS : 98933484, N : 27, type : 1, instance : 0, mul : 938, SampleUS : 0, smp_rate : 1001.12475586}
2017-09-22 09:33:08.14: ISBD {TimeUS : 98933499, N : 27, seqno : 0, x : array('h', [3, 2, 1, 0, 0, 1, 0, 2, 2, 3, 3, 2, 1, 1, 1, 2, 2, 1, 0, 0, 1, 1, 2, 1, 1, 1, 0, 0, 0, 0, 1, 1]), y : array('h', [-1, 0, 0, -1, 0, 0, 1, 1, 2, 1, 0, 0, 0, -1, -1, 0, 1, 0, 0, 0, -1, -1, -1, -1, 0, 0, 0, 0, 0, 0, 2, 1]), z : array('h', [0, 1, 2, 1, 0, 0, -1, -1, -1, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])}
2017-09-22 09:33:08.14: ISBD {TimeUS : 98933542, N : 27, seqno : 1, x : array('h', [0, 0, 0, 1, 2, 4, 5, 3, 0, 0, 0, 2, 2, 1, 2, 2, 1, 0, 0, 1, 1, 0, 0, 2, 3, 3, 2, 2, 2, 1, 1, 1]), y : array('h', [0, 0, 0, 0, 0, -2, -2, 0, 0, 0, 0, 1, 2, 2, 1, 1, 0, 0, 0, 0, 0, 2, 2, 2, 0, 0, 0, 0, 0, 1, 0, 0]), z : array('h', [0, 0, 0, -1, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])}
2017-09-22 09:33:08.25: ISBD {TimeUS : 99044768, N : 27, seqno : 2, x : array('h', [1, 0, 1, 2, 2, 2, 2, 1, 0, 2, 2, 3, 2, 2, 3, 2, 1, 1, 1, 1, 1, 0, 0, 1, 1, 1, 3, 4, 2, 0, 1, 3]), y : array('h', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, -1, -2, -1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0]), z : array('h', [0, 0, 0, 0, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, -1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0])}
2017-09-22 09:33:08.25: ISBD {TimeUS : 99044832, N : 27, seqno : 3, x : array('h', [3, 0, 0, 0, 0, 1, 2, 1, 0, 0, 0, 1, 2, 0, 0, 0, 2, 2, 1, 2, 2, 2, -1, -2, 2, 5, 8, 9, 3, -2, -3, -1]), y : array('h', [0, 0, 0, 0, 0, 0, -1, 0, 0, 0, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 2, 1, 0, -2, 0, 2, 1, -3, -6]), z : array('h', [0, 0, 0, 0, -1, -1, -2, -2, -2, -2, -1, 0, 1, 1, 0, 0, 2, 1, 0, 0, 2, 1, -4, -4, 0, -6, -1, 8, 2, -2, 0, 0])}
```

`mavfft_isb.py` (https://github.com/ArduPilot/pymavlink/blob/master/tools/mavfft_isb.py) interprets these messages to produce pretty pictures:

[image]
[image]

Note that this tool takes the average of all of the per-sensor/instance FFTs to produce the graphs, however animating each sensor/instance, or producing overlays on e.g. a motor-output graph might be other good uses for this data.

meee1, 2017-10-04:
added support here https://github.com/ArduPilot/MissionPlanner/commit/7151df4cd99fceec0505b37fe3646680b71a01b5

tridge, 2017-10-04:
thanks Michael!

closed 2017-10-09 by meee1