ArduPilot/MissionPlanner #1648: Corridor mapping - linear structures mapping

FABIO-CARNEIRO-LOBO, 2017-09-25:
#### Issue details
The proposal is to carry out mapping along linear structures: highways, ducts, railways, etc.

The idea is to (1) indicate an alignment to be mapped (the axis of the linear structure); (2) indicate the mapping width along this axis; (3) collect the photos with percentage coverage values (laeral and sequential).

This question is open on two other related topics:

1.  https://discuss.ardupilot.org/t/corridor-mapping/21398
2.  https://github.com/mavlink/qgroundcontrol/issues/5695

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Flying wing / hexacopter

#### Hardware type
Pixhawk (on my case)

meee1, 2017-09-25:
MP beta already has this
[image]

FABIO-CARNEIRO-LOBO, 2017-09-25:
Uhuuuuu!!!!

nfguide, 2017-09-25:
@meee1  downloaded the latest beta, 1.3.49.29.  Not real clear or evident what the secret decoder is needed to create a corridor mission.  What is the secret?  Is there a document anywhere around on how to plan a corridor mission?

Thanks!

meee1, 2017-09-25:
its a tick box in the normal survey grid screen

nfguide, 2017-09-25:
Great, and thanks.

Quick question.  This is based on a polygon, correct?  If so, what would be your suggestion for a corridor that is 60m in width and 200+km in length?  A polygon doesn't seem reasonable does it?

meee1, 2017-09-25:
that comes down to your flight frame specs. if you have to split it up that's your call.

the closing leg of the polygon is not used. try it and see is the easiest option

nfguide, 2017-09-25:
Yeah, not really working for what is needed.  Great attempt though.  Look forward to what the future holds and may need to look at other options.

Really need to be able to select a single line, put in the corridor width and have it calculate the mission from there.  For example, attempting create a 60m wide polygon in 30km length segments is really not feasible.

jmachuca77, 2017-09-26:
That’s what this does. The first point of the “polygon” is the start of the line. Tinge last point you click will be the end of the line. Then you set the width of the corridor on the settings. And the flight lines generated will be parallel to the perimeter of the polygon. With out taking into account the closing line of the polygon.

Best regards,

Jaime Machuca
CEO | Element Aero | www.element.aero
Mail. jaime@element.aero

Off. +52 (33) 3805 4627
Cel. +52 1 (33) 3945 3350

Este correo electrónico y cualquier archivo transmitido en él, son confidenciales y para uso exclusivo de los individuos y entidades a quienes está dirigido. Si usted no es el destinatario previsto o la persona encargada de recibirlo, y tiene por error este mensaje, queda prohibido y sin validez el uso, difusión, re-envío, reimpresión o copia. Toda oferta y/o aceptación de propuestas comerciales, celebración de contratos u otros actos tendientes a la adquisición de bienes o servicios, así como el establecimiento de cualquier clase de obligación legal para Element Aero SAPI de CV, deberá confirmarse por escrito firmado autógrafamente por funcionario competente, excepto que se cuente con un contrato vigente que autorice el uso de este medio para tales fines. Si usted recibió este correo por equivocación, favor de notificar inmediatamente por este medio a su remitente, y después borrarlo de su correo

nfguide, 2017-09-26:
Great starting point and proof of concept.

Keep working on it, and if I may make a suggestion, attempting to do this using a polygon is not practical for use in the real world.  It needs to be a line drawn, and set the relevant parameters.

FABIO-CARNEIRO-LOBO, 2017-09-26:
An opportunity for this type of survey is the integration with the design tracings of linear structures - which, as a rule, are in line type.
The integration could come from the importation of a track (in dxf or kml file) and the definition of the width of the flight cover (or the offset of the axis) which, together with the lateral covering of the tracks of the photos, would determine the lines of flight.
Could this implementation be possible?

FABIO-CARNEIRO-LOBO, 2017-09-26:
The image presents a general mapping scheme, with a fixed wing equipment - with 80/45% overlapping photos
[image]

nfguide, 2017-09-27:
@FABIO-CARNEIRO-LOBO ,

YOU nailed it, EXACTLY!!!

cczeets, 2017-11-01:
@meee1 Could you add a twist to the heading hold ability in this AWESOME tool you created? I would propose that you have an option where the heading hold maintains parallel to each leg of the corridor without having to spin around 180-degrees at the end of each corridor line.  This would be ideal when using a multi-rotor.

Best,
Steve

FABIO-CARNEIRO-LOBO, 2018-03-25:
@meee1, do you have any suggestion on how the corridor mapping implementation could be based on a linear input, instead of the polygonal one?

meee1, 2018-03-25:
what you are asking already exists, although there is one extra line drawn at the moment.

just draw the line using the polygon tool, and ignore the fact that it draws the final line to home, as the tool doesn't use it anyway

meee1, 2018-03-25:
[image]

FABIO-CARNEIRO-LOBO, 2018-03-25:
Thank's

FABIO-CARNEIRO-LOBO, 2018-04-13:
Could the polygon be inserted into the mission planner by an external kml or shapefile?
Or the user must draw the polygon on the mission planner?

aspirits, 2018-04-30:
@meee1, any chance you could add the ability to have the flight lines on just one side of the polyline/corridor? How it'd be used is in mountainous terrain, you could draw the polyline along the ridge and the resulting flight lines would contour down in order from top to bottom. It would eliminate turning into rising terrain and would be awesome to have in complex terrain where aspect and elevation changes pretty frequent. 

Thanks.

FABIO-CARNEIRO-LOBO, 2018-10-07:
> This corridor option is great and works as one could expect and we have
> used it on multiple occasion! Although I have one question why is it not
> possible to have even number of flight legs? It is always an odd number
> (3,5,7 and so on). In practice it is on many occasion better to have even
> numbers, let say you start the flight on a straight corridor mapping and
> the drone always finishes the mission the furthest away so you either need
> to land there or fly it home when that flight home could be used for the
> mapping as well. Is there any reasoning behind this odd number of flight
> legs that I am not getting?

Maybe it's because the first thing that the corridor mapping do is: set the
principal lane as the middle... and all others will be buffered around it.

Does anyone use the terrain follow altitude on plane mapping?

Tryggvistef, 2018-10-08:
@FABIO-CARNEIRO-LOBO that makes sense and is propably the reason, if thats the case it would be nice to change the method. To answer your earlier question, yes you can insert polygons at least from a shapefile. To do so in MIssion planner you need to right click on the map in the FLIGHT PLAN tab -> expand the Draw Polygon and select from SHP.

FABIO-CARNEIRO-LOBO, 2018-10-08:
Thanks for the answer!