ArduPilot/MissionPlanner #1650: Mission Planner not building

lakshaydang, 2017-09-26:
#### Issue details
I have tried building the mission planner following the steps mentioned in the guide but it is not building. Not even files are loading in Visual studio. I have tried all VS 2013, VS 2015, VS 2017 but all having the same problems. Errors are below. I am stucked. Help me out !!

####Errors coming
1. Error	65	The default XML namespace of the project must be the MSBuild XML namespace. If the project is authored in the MSBuild 2003 format, please add xmlns="http://schemas.microsoft.com/developer/msbuild/2003" to the <Project> element. If the project has been authored in the old 1.0 or 1.2 format, please convert it to MSBuild 2003 format.	C:\Users\Desktop\MP\MissionPlanner\ExtLibs\Comms\MissionPlanner.Comms.csproj	1	1	TrackerHomeGPS

2. Error	18	The attribute "Version" in element <PackageReference> is unrecognized.	C:\Users\Desktop\MP\MissionPlanner\ExtLibs\Comms\MissionPlanner.Comms.csproj	14	41	MissionPlanner.Stats

3. Error	19	The attribute "Update" in element <Compile> is unrecognized.	C:\Users\Desktop\MP\MissionPlanner\ExtLibs\Controls\MissionPlanner.Controls.csproj	37	14	MissionPlanner.Stats

4. Error -	20	The attribute "Remove" in element <Compile> is unrecognized.	C:\Users\Desktop\MP\MissionPlanner\ExtLibs\Mavlink\MAVLink.csproj	18	14	TrackerHomeGPS
 

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2017-09-26:
upgrade your VS to 2017.3-4

closed 2017-09-26 by meee1

ddomit, 2018-04-24:
How do you add the missing references? I cant find a way to import a whole folder with the dependencies from the installed mission planner (The documentation says to do this, I dont know if its still right)