ArduPilot/MissionPlanner #1666: Multiple load fails when i try to open MissionPlanner.sln

brunolsantos, 2017-10-16:
#### Issue details
_I just opened the project and several files failed to load, i tried to add `xmlns="http://schemas.microsoft.com/developer/msbuild/2003"` to xml but the problem continues.
Is it a local error or something in Mission Planner?_

#### Version
_1.3.50_

#### Logs
_C:\Users\xxxx\Documents\Projects\MissionPlanner\mission-planner-dev\MissionPlanner-MissionPlanner1.3.50\ExtLibs\Arduino\Arduino.csproj : error  : The default XML namespace of the project must be the MSBuild XML namespace. If the project is authored in the MSBuild 2003 format, please add xmlns="http://schemas.microsoft.com/developer/msbuild/2003" to the <Project> element. If the project has been authored in the old 1.0 or 1.2 format, please convert it to MSBuild 2003 format.  C:\Users\xxxx\Documents\Projects\MissionPlanner_aerials\mission-planner-dev\MissionPlanner-MissionPlanner1.3.50\ExtLibs\Arduino\Arduino.csproj_

Naphier, 2017-10-17:
See:
https://github.com/ArduPilot/MissionPlanner/issues/1665

You need VS 2017 and you need to install .net standard 2.0

I still have some warnings of missing namespaces, but it compiles and runs fine now.

brunolsantos, 2017-10-17:
Wow Nice, i was reading your issue earlier, thanks for the reply!!

brunolsantos, 2017-10-17:
It worked, thanks!!!  One final question, i remembered that all windows forms were located at GCSViews but now the developers changed to somewhere else, do you know where are the views?

closed 2017-10-17 by brunolsantos