ArduPilot/MissionPlanner #1669: Failed to start the application System.Speech.dll

ruyi7952, 2017-10-19:
#### Issue details
Failed to start the application
#### Version
1.3.50 msi
1.3.50 zip
#### Logs
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
INFO Program - ******************* Logging Configured *******************
INFO Program - 64bit os True, 64bit process True
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883_OLD or 1
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883 or 7
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype 17 or DEVTYPE_ACC_LSM303D
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 36 or DEVTYPE_GYR_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 4 devtype 34 or DEVTYPE_GYR_L3GD20
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 36 or DEVTYPE_GYR_MPU9250
FATAL Program - Fatal app exception
System.IO.FileLoadException: 未能加载文件或程序集“System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35”或它的某一个依赖项。找到的程序集清单定义与程序集引用不匹配。 (异常来自 HRESULT:0x80131040)
文件名:“System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35” ---> System.IO.FileLoadException: 未能加载文件或程序集“System.Speech, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35”或它的某一个依赖项。找到的程序集清单定义与程序集引用不匹配。 (异常来自 HRESULT:0x80131040)
文件名:“System.Speech, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35”

警告: 程序集绑定日志记录被关闭。
要启用程序集绑定失败日志记录，请将注册表值 [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD)设置为 1。
注意: 会有一些与程序集绑定失败日志记录关联的性能损失。
要关闭此功能，请移除注册表值 [HKLM\Software\Microsoft\Fusion!EnableLog]。

   在 MissionPlanner.MainV2..ctor()
   在 MissionPlanner.Program.Main(String[] args) 位置 C:\Users\michael\Source\Repos\MissionPlanner\Program.cs:行号 221

System.IO.FileLoadException: 未能加载文件或程序集“System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35”或它的某一个依赖项。找到的程序集清单定义与程序集引用不匹配。 (异常来自 HRESULT:0x80131040)
文件名:“System.Speech, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35” ---> System.IO.FileLoadException: 未能加载文件或程序集“System.Speech, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35”或它的某一个依赖项。找到的程序集清单定义与程序集引用不匹配。 (异常来自 HRESULT:0x80131040)
文件名:“System.Speech, Version=3.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35”

警告: 程序集绑定日志记录被关闭。
要启用程序集绑定失败日志记录，请将注册表值 [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD)设置为 1。
注意: 会有一些与程序集绑定失败日志记录关联的性能损失。
要关闭此功能，请移除注册表值 [HKLM\Software\Microsoft\Fusion!EnableLog]。

   在 MissionPlanner.MainV2..ctor()
   在 MissionPlanner.Program.Main(String[] args) 位置 C:\Users\michael\Source\Repos\MissionPlanner\Program.cs:行号 221

Press any key to exit!

OlivierJB, 2017-12-05:
Have had similar issue, error log below, after changing baudrate in connect dialog box.
Fix for now: Either delete  MP config.xml or edit it and look for any baud rate that is non standard (just set em all to 57600 will work). Config is typically in My Documents -> Mission Planner

Thanks to @WickedShell for that!

------------------
```
_If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
INFO Program - ******************* Logging Configured *******************
INFO Program - 64bit os True, 64bit process True
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883_OLD or 1
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883 or 7
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype 17 or DEVTYPE_ACC_LSM303D
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 36 or DEVTYPE_GYR_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 4 devtype 34 or DEVTYPE_GYR_L3GD20
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 36 or DEVTYPE_GYR_MPU9250
INFO MissionPlanner.MainV2 - Mainv2 ctor
INFO MissionPlanner.MainV2 - Loading config
INFO MissionPlanner.srtm - .cctor
INFO MissionPlanner.srtm - C:\ProgramData\Mission Planner\srtm
INFO MissionPlanner.Utilities.Speech - TTS: init, mono = False
FATAL Program - Fatal app exception
System.FormatException: Input string was not in a correct format.
   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseInt32(String s, NumberStyles style, NumberFormatInfo info)
   at MissionPlanner.MainV2.CMB_baudrate_TextChanged(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnTextChanged(EventArgs e)
   at System.Windows.Forms.ComboBox.OnTextChanged(EventArgs e)
   at System.Windows.Forms.Control.set_Text(String value)
   at System.Windows.Forms.ComboBox.set_Text(String value)
   at System.Windows.Forms.ComboBox.UpdateText()
   at System.Windows.Forms.ComboBox.set_SelectedIndex(Int32 value)
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Main(String[] args)
System.FormatException: Input string was not in a correct format.
   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseInt32(String s, NumberStyles style, NumberFormatInfo info)
   at MissionPlanner.MainV2.CMB_baudrate_TextChanged(Object sender, EventArgs e)
   at System.Windows.Forms.Control.OnTextChanged(EventArgs e)
   at System.Windows.Forms.ComboBox.OnTextChanged(EventArgs e)
   at System.Windows.Forms.Control.set_Text(String value)
   at System.Windows.Forms.ComboBox.set_Text(String value)
   at System.Windows.Forms.ComboBox.UpdateText()
   at System.Windows.Forms.ComboBox.set_SelectedIndex(Int32 value)
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Main(String[] args)

Press any key to exit!_

```

msrks, 2017-12-11:
I fell into the same situation, but I cured it by deleting config.xml. Thanks @OlivierJB !

meee1, 2017-12-11:
latest MP version should resolve the second issue here. as the original issue is 100% unrelated

closed 2017-12-11 by meee1

Kirilpython, 2018-11-26:
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
Data Dir C:\ProgramData\Mission Planner\
Log Dir C:\Users\UAV-LAB\Documents\Mission Planner\logs
Running Dir C:\Users\UAV-LAB\Desktop\MissionPlanner-master\bin\Debug\net461\
User Data Dir C:\Users\UAV-LAB\Documents\Mission Planner\
INFO Program - ******************* Logging Configured *******************
INFO Program - 64bit os True, 64bit process True
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883_OLD or 1
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_I2C bus 1 address (i2c addr or spi CS) 30 devtype DEVTYPE_HMC5883 or 7
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype DEVTYPE_LSM303D or 2
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype DEVTYPE_AK8963 or 4
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 3 devtype 17 or DEVTYPE_ACC_LSM303D
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 22 or DEVTYPE_ACC_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 1 devtype 36 or DEVTYPE_GYR_MPU9250
bus type BUS_TYPE_SPI bus 4 address (i2c addr or spi CS) 4 devtype 34 or DEVTYPE_GYR_L3GD20
bus type BUS_TYPE_SPI bus 1 address (i2c addr or spi CS) 4 devtype 36 or DEVTYPE_GYR_MPU9250
FATAL Program - Fatal app exception
System.TypeInitializationException: The type initializer for 'MissionPlanner.MainV2' threw an exception. ---> System.IO.FileLoadException: Could not load file or assembly 'SvgNet, Version=1.0.0.0, Culture=neutral, PublicKeyToken=38326cb7e06851fc' or one of its dependencies. Strong name validation failed. (Exception from HRESULT: 0x8013141A) ---> System.Security.SecurityException: Strong name validation failed. (Exception from HRESULT: 0x8013141A)
   --- End of inner exception stack trace ---
   at MissionPlanner.MAVState..ctor(MAVLinkInterface mavLinkInterface, Byte sysid, Byte compid)
   at MissionPlanner.Mavlink.MAVList..ctor(MAVLinkInterface mavLinkInterface)
   at MissionPlanner.MAVLinkInterface..ctor()
   at MissionPlanner.MainV2..cctor()
   --- End of inner exception stack trace ---
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Main(String[] args)
System.TypeInitializationException: The type initializer for 'MissionPlanner.MainV2' threw an exception. ---> System.IO.FileLoadException: Could not load file or assembly 'SvgNet, Version=1.0.0.0, Culture=neutral, PublicKeyToken=38326cb7e06851fc' or one of its dependencies. Strong name validation failed. (Exception from HRESULT: 0x8013141A) ---> System.Security.SecurityException: Strong name validation failed. (Exception from HRESULT: 0x8013141A)
   --- End of inner exception stack trace ---
   at MissionPlanner.MAVState..ctor(MAVLinkInterface mavLinkInterface, Byte sysid, Byte compid)
   at MissionPlanner.Mavlink.MAVList..ctor(MAVLinkInterface mavLinkInterface)
   at MissionPlanner.MAVLinkInterface..ctor()
   at MissionPlanner.MainV2..cctor()
   --- End of inner exception stack trace ---
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Main(String[] args)

Press any key to exit!

please help me

meee1, 2018-11-26:
you will need to reinstall the latest version. either use the MSI, or use the installer.exe file in the mission planner directory

meee1, 2018-11-26:
turn off assembly signing on 'SvgNet

Kirilpython, 2018-11-26:
how?

Kirilpython, 2018-11-26:
you have team viewer

Kirilpython, 2018-11-26:
?please reply