ArduPilot/MissionPlanner #1672: Flight planner random path feature request

DanielOBenson, 2017-10-23:
I have developed an autonomous lawnmower.  Flight paths using the various options in the Flight Plan tab of Mission Planner miss some grass.
 
I think It would be nice to have an option to conduct a flight path using a i-robot type of pattern.  This would drive the robot on a random(1,360) heading for a random amount of  time=  random(time minimum, time maximum).

When the lawnmower comes to the geofence its heading would change to the compliment of present heading.

Should you think this is a worthwhile addition to  feature rich Mission Planner I would be happy to pay for your efforts.

Thank you in advance.

magicrub, 2017-10-23:
I think this is actually a feature of the autopilot, not the GCS.

DanielOBenson, 2017-10-23:
Sorry to bother you.  Thank you.