ArduPilot/MissionPlanner #1677: geo ref images using time offset - incorrect date time inforamtion exif data is used

CharlieGUS, 2017-10-30:
When using geo-ref images time offset geo tagging function, the wrong data is used from the image exif - data/time appears as 0001:01:01 00:00:00 

to reproduce:

open geo-ref images, load any log and any set of images. Select time offset and hit estimate offset. Observe something like this:

1 Picture GOPR8717 with DateTime: 0001:01:01 00:00:00
1 GPS Log Msg: 2017:10:27 17:39:21
1 Est: -63644722761.6

The gopro date and time exif information was correct in this example, I've checked with images from other cameras, etc. as well.

Versions tested
1.3.50
1.3.50.3

jmachuca77, 2017-10-31:
I ran into this problem also, I can confirm it.

meee1, 2017-11-05:
Can someone posted one of the images with the issue

jmachuca77, 2017-11-06:
Sent a log and images via email on a dropbox link

Best regards,

Jaime Machuca
CEO | Element Aero | www.element.aero <http://www.element.aero/>
Mail. jaime@element.aero <mailto:jaime@element.aero>

Off. +52 (33) 3805 4627
Cel. +52 1 (33) 3945 3350

Este correo electrónico y cualquier archivo transmitido en él, son confidenciales y para uso exclusivo de los individuos y entidades a quienes está dirigido. Si usted no es el destinatario previsto o la persona encargada de recibirlo, y tiene por error este mensaje, queda prohibido y sin validez el uso, difusión, re-envío, reimpresión o copia. Toda oferta y/o aceptación de propuestas comerciales, celebración de contratos u otros actos tendientes a la adquisición de bienes o servicios, así como el establecimiento de cualquier clase de obligación legal para Element Aero SAPI de CV, deberá confirmarse por escrito firmado autógrafamente por funcionario competente, excepto que se cuente con un contrato vigente que autorice el uso de este medio para tales fines. Si usted recibió este correo por equivocación, favor de notificar inmediatamente por este medio a su remitente, y después borrarlo de su correo

meee1, 2017-11-06:
new beta uploaded. hopefully fixs this bug